dl9rdz/rdz_ttgo_sonde #285: Screen Sometimes White when Starting

WaldperlachFabi, 2022-03-25:
**Describe the bug**
The Display Is sometimes white when starting and the rdzttgo Programm doesn’t start

**To Reproduce**
Steps to reproduce the behavior:
Turn it on, sometimes it‘ll work normally, sometimes it‘ll be white
**Expected behavior**
Run the ttgo rdz side programm 

**Screenshots**
//

**Version and specific configuration:**
 - DollaTek T-Beam 433MHz ESP32 WiFi Drahtloses Bluetooth Modul GPS NEO-6M SMA LORA 18650

 - ILI9225 LCD-TFT-Anzeigemodul für serielle Schnittstelle, serielle periphere Schnittstelle 2,0 Zoll 3V / 5,5V 176x220

**Additional context**
- devel20220123-B16

dl9rdz, 2022-03-26:
This seems to be an hardware issue.

If you have connected the display's VCC pin to 3.3V on your TTGO, and your display has something labelled J1 on the back, then you should short-circuit (bridge) these two pins. Did you do that already?

WaldperlachFabi, 2022-03-26:
Yea I did that

closed 2022-12-31 by WaldperlachFabi