dl9rdz/rdz_ttgo_sonde #288: RS92 is not decoded properly

mmadela, 2022-03-30:
today an RS92 sonde is flying nearby, but it is not decoded properly,
only its id is parsed
I am putting the log output from serial port, I hope you can do something with it

Sonde::setup() done: Type RS92 Freq 405.300000, AFC BW: 12500, RX BW: 12500
RS92::receive() start at 16333630
Status form: size=7008 bytes
rs corr is -1 --- data:0x3ffbe820 data1:0x3ffbe820 data2:0x3ffbe620 lastframe=1
RS92::receive(): new frame complete after 981ms
Sonde::timeoutEvent: RX
Sonde:receive(): result 0 (RX_OK), event 0a => action 3f
RS92::receive() start at 16334638
waitRXcomplete returning 3f00 (RX_OK)
decoding frame 1
2a2a2a2a2a1065106d1d20204631313234393534006100453a25a872503e2842c4574429e903b8c2e73d690c36d00f7fb40fa3b50f6b43116ab5116ac2107f7c0dd97d0df0ad673d95ac49cdb16492b9b313c41616279f6ccf05099699b4eadbf87a4adafa6949b91958ea580ad89d49e97a584a084c4b2ba94bfa0b220d6cb54f79b2bbe1bd12aaf385414132eb6195036feed5f94de1052eaafdaa88ad188a686aeceeefefafcb6de90be9efefef0b0f8b8da86b08ab0868086b08aa0c088808a8eaeb4f488848bea1e5b097ab07afbfbd9595b5f1aba6bb8f98b7b7ae32b3b192b1726d5fae24a18387a43a313767
GPS time is fff30487 (-850809)
[ 7533] (F1124954) sA -20:-20:-9.707
Setting display mode 1
setLayout: 1 (max is 8)
MAIN: loopDecoder: action 3f (Default Display) => 255  [current: main=19, rxtask=19]
next sondehub frequncy import in 188 seconds
mon2raw for N0CALL>APRRDZ:;F1124954 *234532h0000.00N/00000.00EO!w!!!Clb=0.0m/s 405.300MHz Type=RS92 rdzTTGOsonde
Sending AXUDP
MAIN: updateDisplay started
drawRSSI: 6 0 103 (19)[4]
MAIN: updateDisplay done (after 94 ms)

or

Sonde::setup() done: Type RS92 Freq 405.300000, AFC BW: 12500, RX BW: 12500
RS92::receive() start at 16321362
RS92::receive() timed out
Sonde::timeoutEvent: NORX
Sonde:receive(): result 1 (RX_TIMEOUT), event 0b => action 80
RXtask: start DECODER for sonde 0 (was DECODER)
Sonde::setup() start on index 0
Status form: size=7008 bytes
waitRXcomplete returning 8001 (RX_TIMEOUT)
decoding frame 2
2a2a2a2a2a106557edbdb3b7aeb3f333b2b1b2b2bfa6bfbcb0b695a18bbdb5af8eaaaeb585bf918fb8a1a5fc9789fe97838e55b44fd3a19bdc0e02a53d86a4bf6d1c97a88074d585724245757798f9e46766851600af1f5f3f3f778f6fcf3f9800f185ad018de99509185ec6ff99398c038aac230096cf8d09d4c4280078d897084b36feff56c0a702aaf7b500167fa55d403f657f5d762f791831677f4b2a0e7f2b14647f572134202c61697f3618317640002040286a4c7f000004400b47357f17254b7d7f7e0f6d59397f7e7f7f6947007e7e7f7e7f6145b34f69d4a1e692e483eee1adc4c7cdf0f7174c7b78136a
GPS time is 1e0571 (1967473)
MAIN: loopDecoder: action 80 (Sonde=0) => 65  [current: main=0, rxtask=0]
next sondehub frequncy import in 198 seconds
MAIN: updateDisplay started
MAIN: updateDisplay done (after 67 ms)

hr-ru, 2022-03-30:
Seems like the ftp server used by the default config for the eph data is offline, and without that, no position decoding.

This could be an server you could use instead:  gssc.esa.int/gnss/data/daily/

Requires a reboot to fetch fresh eph data

oe3jtb, 2022-03-30:
Rinex data up2date?

mmadela, 2022-03-30:
the About displays;
RS92 RINEX eph state: Disabled

and the config displays for FTP server for ephemeris data (RS92 decoder):
igs.bkg.bund.de/IGS/BRDC/

Does it mean rs92 does not work out of the box?

mmadela, 2022-03-30:
ok, I've entered the [gssc.esa.int/gnss/data/daily/](url)
I hope it will work next time or should I change something more?
(the frequency for the sonde was received automatically _by SondeHub frequency import_)

hr-ru, 2022-03-30:
That should be fine (but I have no influence on external ftp server operators. more and more people switch to https only, which is not supported by the ttgo software)

One important thing is that rinex data is downloaded only if a frequency with mode RS92 is active on startup.
So you have to enable such afrequency in the QRG tab first, and then do a reset.

mmadela, 2022-03-30:
so in other words it will not work, if the RS92 is received by SondeHub import?
(I guess yes, I've took a look in source)
And it is mandatory to have a RS92 placeholder on the QRG list?
Could it be changed in future to work without this?

hr-ru, 2022-03-30:
Not automatically.

It is sufficient to have a inactive line in the QRG list with type RS92. 

More important is that you need fresh (like not older than mayby 6h) eph data, so you should reboot periodically. As more and more people are using the TTGO running 24/7, it might make sense to automated that in the future...

dl9rdz, 2022-03-31:
That may be the case, I will have to check.

The RS92 was mainly tested by me back when the TTGO was only used for mobile decoding without any feeds to Internet, and for that I assumed back then that seeing anything on the display, even if partially wrong, is better than seeing nothing. With the integration of SondeHub uploads, we carefully checked most sonde types for data quality, but maybe I was not that thorough enough with the RS92.