dl9rdz/rdz_ttgo_sonde #290: Data from rdzTTGOSonde not uploaded to server

ivanpiter, 2022-04-08:
Please, look here:
https://github.com/projecthorus/sondehub-tracker/issues/255

Earlier same day data from another sonde was uploaded and seen on sondehub.org
But later one is not uploaded with good packets received.
Here is the log of not uploaded packets:

MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 129908, unused stack: 3800
[RSSI=220 FEI=-2441 AFC=1953] CFG(1):1CB1557 DAT(1):2301715F10A92 DAT(1):10E1B6BE2C753 1c:i=6,top=15meas 1 is 177.333740 (726359,12)
DAT[2]: GPS-lat: 58.73, hor-V: 42.65 DAT[3]: GPS-lon: 28.32, dir: 113.81
[RSSI=217 FEI=-2319 AFC=1953] CFG(1):28A2EAD DAT(0):000E00D4F9164 DAT(0):F8AF02FC04355 28:i=13,top=15meas 2 is 2606.675781 (667309,8)
DAT[4]: GPS-height: 9177.16, vv: -17.70 DAT[5]: (?)
[RSSI=218 FEI=-305 AFC=1953] CFG(0):8045560 DAT(3):80A0A280A3806 DAT(2):A49E98A3A39E7 80:i=4,top=15 DAT[6]: (?) DAT[7]: (?)
[RSSI=218 FEI=-1525 AFC=1953] CFG(0):086689C DAT(1):7E6433A107008 DAT(1):000002AC02040 08:i=14,top=15meas 0 is 1640.609375 (419996,8)
DAT[8]: Date: 2022-04-06 14:33z Cycle done: good is 1ff
Meas: 1640.609375 102.667480 1115.183594
Temp: -56.390495
DAT[0]: Packet counter: 172
Sonde:receive(): result 0 (RX_OK), event 00 => action ff
waitRXcomplete returning ff00 (RX_OK)
mon2raw for N0CALL>APRRDZ:;D636556 *143305h5843.78N/02819.36EO115/083/A=030109!w8a!Clb=-17.7m/s t=-56.4C 403.250MHz Type=DFM ser=636556 rdzTTGOsonde
Sending AXUDP
sondehub_send_data()
shState = 3
21a
, {"software_name": "rdzTTGOsonde","software_version": "master_v0.9.1","uploader_callsign": "DG_RDZTTGO","time_received": "2022-04-06T14:33:05.000Z","manufacturer": "Graw","serial": "636556","datetime": "2022-04-06T14:33:05.000Z","lat": 58.72971,"lon": 28.32278,"alt": 9177.16016,"frequency": 403.250,"vel_h": 42.65000,"vel_v": -17.70000,"heading": 113.81000,"rssi": -109.5,"frame": 1333290785,"type": "DFM","sats": 7,"subtype": "DFM09","temp": -56.4,"batt": 4.59,"uploader_antenna": "Dummy","uploader_position": [59.730400,30.388300,45]}
MAIN: updateDisplay started
drawRSSI: 6 0 109 (14)[4]
MAIN: updateDisplay done (after 89 ms)

MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 129908, unused stack: 3800
[RSSI=219 FEI=-183 AFC=1953] CFG(1):1CB1568 DAT(0):160A801017701 DAT(0):23016B6E10BA2 1c:i=6,top=15meas 1 is 177.337891 (726376,12)
DAT[1]: UTC-msec: 6000 DAT[2]: GPS-lat: 58.73, hor-V: 42.82
[RSSI=216 FEI=-1342 AFC=1953] CFG(1):28A2E1E DAT(0):10E1D1352C123 DAT(2):000DF9EBF9104 28:i=13,top=15meas 2 is 2606.117188 (667166,8)
DAT[3]: GPS-lon: 28.32, dir: 112.82 DAT[4]: GPS-height: 9159.47, vv: -17.76
[RSSI=218 FEI=-1525 AFC=1953] CFG(0):9000000 DAT(0):F8AF02FB042F5 DAT(2):475B454950406 90:i=5,top=15 ID:90[0] CNT:233,0,st=144,lastfrid=172 DAT[5]: (?) DAT[6]: (?)
[RSSI=219 FEI=-1342 AFC=1953] CFG(1):0866999 DAT(0):5254595A5D5F7 DAT(0):7E6433A107008 08:i=14,top=15meas 0 is 1641.597656 (420249,8)
DAT[7]: (?) DAT[8]: Date: 2022-04-06 14:33z Cycle done: good is 1ff
Meas: 1641.597656 102.667480 1115.183594
Temp: -56.399757

Sonde:receive(): result 0 (RX_OK), event 00 => action ff
waitRXcomplete returning ff00 (RX_OK)
mon2raw for N0CALL>APRRDZ:;D636556 *143306h5843.77N/02819.40EO114/083/A=030051!wBf!Clb=-17.8m/s t=-56.4C 403.250MHz Type=DFM ser=636556 rdzTTGOsonde
Sending AXUDP
sondehub_send_data()
shState = 3
21a
, {"software_name": "rdzTTGOsonde","software_version": "master_v0.9.1","uploader_callsign": "DG_RDZTTGO","time_received": "2022-04-06T14:33:06.000Z","manufacturer": "Graw","serial": "636556","datetime": "2022-04-06T14:33:06.000Z","lat": 58.72956,"lon": 28.32346,"alt": 9159.46973,"frequency": 403.250,"vel_h": 42.82000,"vel_v": -17.76000,"heading": 112.82000,"rssi": -111.0,"frame": 1333290786,"type": "DFM","sats": 7,"subtype": "DFM09","temp": -56.4,"batt": 4.59,"uploader_antenna": "Dummy","uploader_position": [59.730400,30.388300,45]}
MAIN: updateDisplay started
drawRSSI: 6 0 111 (14)[4]
MAIN: updateDisplay done (after 89 ms)

MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 129908, unused stack: 3800
[RSSI=222 FEI=-2441 AFC=1953] CFG(1):1CB1569 DAT(1):000002AD02040 DAT(2):160A80101B581 1c:i=6,top=15meas 1 is 177.338135 (726377,12)
DAT[0]: Packet counter: 173 DAT[1]: UTC-msec: 7000
[RSSI=219 FEI=-1770 AFC=1953] CFG(0):28A2D24 DAT(0):230165A910BD2 DAT(0):10E1EBE22BE13 28:i=13,top=15meas 2 is 2605.140625 (666916,8)
DAT[2]: GPS-lat: 58.73, hor-V: 42.85 DAT[3]: GPS-lon: 28.32, dir: 112.33
[RSSI=216 FEI=-1525 AFC=1953] CFG(1):AC00090 DAT(2):000DF306F9124 DAT(0):F8B002FA04295 ac:i=0,top=15 ID:AC[0] CNT:120,116,st=172,lastfrid=172 ID OK DAT[4]: GPS-height: 9141.82, vv: -17.74 DAT[5]: (?)
[RSSI=217 FEI=-1464 AFC=1953] CFG(2):0866632 DAT(0):80A0A280A3806 DAT(0):A49E99A3A39E7 08:i=14,top=15meas 0 is 1638.195312 (419378,8)
DAT[6]: (?) DAT[7]: (?)
[RSSI=221 FEI=-2380 AFC=1953] CFG(0):1CB1578 DAT(0):7E6433A107008 DAT(0):000002AE02040 1c:i=6,top=15meas 1 is 177.341797 (726392,12)
DAT[8]: Date: 2022-04-06 14:33z Cycle done: good is 1ff
Meas: 1638.195312 102.667480 1115.183594
Temp: -56.367851
DAT[0]: Packet counter: 174
Sonde:receive(): result 0 (RX_OK), event 00 => action ff
waitRXcomplete returning ff00 (RX_OK)
mon2raw for N0CALL>APRRDZ:;D636556 *143307h5843.76N/02819.44EO113/083/A=029993!wNo!Clb=-17.7m/s t=-56.4C 403.250MHz Type=DFM ser=636556 rdzTTGOsonde
Sending AXUDP
sondehub_send_data()
shState = 3
21a
, {"software_name": "rdzTTGOsonde","software_version": "master_v0.9.1","uploader_callsign": "DG_RDZTTGO","time_received": "2022-04-06T14:33:07.000Z","manufacturer": "Graw","serial": "636556","datetime": "2022-04-06T14:33:07.000Z","lat": 58.72942,"lon": 28.32414,"alt": 9141.82031,"frequency": 403.250,"vel_h": 42.85000,"vel_v": -17.74000,"heading": 112.33000,"rssi": -110.5,"frame": 1333290787,"type": "DFM","sats": 7,"subtype": "DFM09","temp": -56.4,"batt": 4.59,"uploader_antenna": "Dummy","uploader_position": [59.730400,30.388300,45]}
1
]
0

MAIN: updateDisplay started
drawRSSI: 6 0 110 (14)[4]
MAIN: updateDisplay done (after 88 ms)

dl9rdz, 2022-04-08:
So this is about the "however it appears the rdz_ttgo_sonde code is not showing the response from sondehub which would be telling you that this is happening."?

Actually, you should see this response. This should, however, be a bit later than the "1 ] 0" part you are seeing. So you have cut off that part... I assume you have more of the serial log?

ivanpiter, 2022-04-08:
I didn't save more logs, sorry. I thought 2 full rounds is enough.
Only can wait for next "not uploading" sonde....
It happens for me first time. Usually everything uploads fine.

dl9rdz, 2024-01-07:
I'm closing this for now.
If this issue re-appears, feel free to re-open with additional log data.

closed 2024-01-07 by dl9rdz