dl9rdz/rdz_ttgo_sonde #295: ttgo v2.16 + ili9225

14ri004, 2022-04-22:
Hello, is it possible to connect an ILI9225 TFT screen to a TTGO v2.16 (the small one) instead of the OLED screen??, thank you

dl9rdz, 2022-04-22:
Sure.

You will have to configure I/O pins manually in the config, no support for automated configuration for that setup.

I suggest you use the same wiring as suggested for the T-BEAM 0.7 in the wiki (which should be the same as the one shown here: https://skp.wodzislaw.pl/wp-content/uploads/2019/03/sch_TFT_1.6.png)

14ri004, 2022-04-23:
Hello, I tested this configuration, but it does not work, the decoding is very difficult, if at all, I tried several connections, but no more result, the screen of my TTGO is down, and I wanted to replace it with an ILI9225, but it doesn't seem to work, waiting for a solution, thanks.

dl9rdz, 2022-04-23:
You are saying "decoding is very difficult", does that mean that display itself is working, or do you see nothing on the display as well?

The ILI9225 displays are known to cause some RF noise. Keeping cables to the display short, away from the antenna, and shielding (TTGO and display within a casing with metal spray or aluminium foil) usually help. 

But in any case, in my experience the effect is not that dramatic as to call it "decoding is very difficult". With outside antenna a few meters away, no noticeable difference. With the antenna directly on the TTGO, its a bit less sensitive on weak signals, but I would call it "good enough" :)

14ri004, 2022-04-23:
yes sometimes it decodes, but if I reboot it, no more decoding, indeed the screen must make RF noise, which interferes with the decoding, it's not a success too bad, the OLED screen is out of order, so I can only use it from WIFI, thanks for your help.

14ri004, 2022-04-23:
if I remove the screen from the TTGO a good distance, it could solve the QRM problem of the screen or put a ferit TOR, I will try and let you know what happens next, thank you

closed 2022-05-30 by 14ri004