dl9rdz/rdz_ttgo_sonde #297: Balloon paths not displaying properly on Sondehub

XionUAV, 2022-04-25:
I have an RDZ tracker setup in Southern Idaho tracking the balloons out of Elko, Nevada. Occasionally on Sondehub it will show two balloons with different serial numbers but only one is live tracking. Then it will display a path out and back for the other balloon as if it was a past flight.

I thought this was an isolated incident until installing another RDZ tracker in Lehi, Utah that occasionally does the same thing. Yesterday I recovered a balloon in Southern Idaho that was launched from Elko, Nevada a few days before and tracked with an RDZ station. When I tried to log the recovery by entering the serial number in Sondehub, the attached screenshot is what I got. It showed a path to the area where I actually recovered the balloon but also a path back to the launch point. Since it shows the balloon at the launch point, I was unable to log the recovery because Sondehub returned an error of "Distance Limit Exceeded." I have never seen or experienced this problem with my Auto_RX stations.

[image]

dl9rdz, 2022-04-25:
Thanks for reporting this. I will have to investigate. I see that you are using the latest master version, and that should have a fix for this issue, but something seems to be going wrong. I can see the following data in the sondehub database:

last valid frame:
2022-04-21T13:46:28.000Z 20041523 1334583988 42.10449 -113.7306 3180.0 N7ZDX-3

and then:
2022-04-21T23:23:00.000Z 21013116 1334618580 40.96049 -115.6911 6592.1001 N7ZDX-3
_2022-04-21T23:23:37.000Z **20041523** 1334618617 40.95634 -115.69623 6447.8999 N7ZDX-3
[...]
2022-04-21T23:23:48.000Z **20041523** 1334618628 40.95835 -115.69376 6519.0 N7ZDX-3_
2022-04-21T23:23:54.000Z 21013116 1334618634 40.95948 -115.69256 6556.3999 N7ZDX-3

This definitely should not happen, and right now I have no explanation for that.

XionUAV, 2022-04-26:
I'm glad you are able to verify that there is an actual issue. Thank you
for responding so quickly. Hopefully you'll be able to find the source of
the issue. Good luck.

dl9rdz, 2022-04-26:
Ok, I have made some enhancements to the DFM ID handling. This might fix the problem (in devel-20220426).
I will release a new master version some time next week after more testing.

XionUAV, 2022-04-26:
Very good. I'll watch for the new version.

XionUAV, 2022-04-26:
I don’t know if this will be of any further value or not but right now an RDZ station I built for a guy in Spokane, Washington is producing the problem I described.

dl9rdz, 2022-04-26:
Yes, thanks, this seems to be the same problem, and fixed in the devel version.

XionUAV, 2022-04-27:
Very good.

XionUAV, 2022-05-06:
Is there any chance the new version is ready to download? I’m building a tracker to send to one of the members of my Radiosonde North America group and there is no way he would be able to load the firmware himself if I loaded the old firmware and sent the tracker to him. Very nice guy and fellow Ham, but severely technically challenged. Thanks.

Mooni123, 2022-05-06:
Hello,

Why are you don't using the latest development build from here:
https://rdzsonde.mooo.com/download.html

XionUAV, 2022-05-06:
Very good. Thank you.

 

The last time we spoke you said you wanted to test your changes for a week or so before releasing a final version so I wasn’t sure if that had happened or not.

 

https://rdzsonde.mooo.com/download.html

dl9rdz, 2022-05-06:
I am travelling right now and did not create the new master version yet.
But I did not hear anything bad about the devel version, so I might just merge it into the master branch. But anyway, the functionality will be exactly the same as in the current devel version, so feel free to use just that for now.

XionUAV, 2022-05-06:
Perfect. Thank you. I’ve loaded the devel version onto the TTGO board and will test it tonight before sending it to the guy tomorrow. I’ll also share this info with my Radiosonde North America Facebook group.

 

By the way, here’s the case I designed and use with the TTGO board and your firmware. The front cover shape is the same used with the MySondy in order to fit a 2000 mAh battery. The back cover is deeper and there are no mounting ears on that one. And of course that one has “GO” on the front.

 

 

Save travels.

XionUAV, 2022-07-07:
Looks like there might still be an issue with the new version of firmware.
However, I only started noticing this after there was a power-bump at the
tracker's location. It wasn't displaying on Sondehub after that. I had a
person there unplug it, wait then plug it back in. However, doing this
obviously shouldn't affect the firmware. The position of the lower balloon
in this screenshot is where the RDZ lost signal. It was tracking the same
balloon.