dl9rdz/rdz_ttgo_sonde #300: Screen off after last dev update

dreamfalcon, 2022-04-27:
After the last dev updated the screen never turns on.
Tried reverting to the master but same problem.
Probably a hardware problem!?

eben80, 2022-05-06:
The green led flashes once during startup?

dreamfalcon, 2022-05-06:
Yes.
The web interface works fine.

eben80, 2022-05-06:
How are you flashing it? With platformIO or the flasher?
It might be an issue with the data files.
Perhaps try a full flash if you're using the flasher or reflash the filesystem if using platformio.

dreamfalcon, 2022-05-07:
Was using OTA.
But now used the esptool and same problem. Probably a harware problem :(
esptool --chip esp32 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 master_v0.9.1-B16-full.bin

eben80, 2022-05-07:
It might we worth erasing the flash and then flashing the original paxcounter software just to make sure it's a hardware issue.

closed 2023-12-03 by dl9rdz