dl9rdz/rdz_ttgo_sonde #303: ILI9341 + TTGO 2.16

14ri004, 2022-05-28:
hello, I connected an ili9341 screen to a TTGO 2.16, on the screen there are not the same pins as on the ili9225, it lights up fine but the TTGO data is not displayed on the screen. screen, which could provide me with the correct wiring between ili9341 and TTGO 2.16, with many thanks in advance.

my wiring:
vcc + led
gnd
reset
cs

nda ????
rs ???
[image]

dl9rdz, 2022-05-28:
Make sure to check https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Supported-displays

SDO(MISO) => Not connected
LED => 3V3
SCK => 13 (SCK/SCL/CLK is the same)
**SDI(MOSI) => 4 (SDA)**
**DC => 2 (that is "RS")**
RESET => 14
CS => 0
GND => obvious
VCC => obvious

Make sure to change display type to ILI9341 in the config AND REDUCE SPI SPEED. The ILI9341 is slower than the ILI9225 wrt max bus speed.

14ri004, 2022-05-28:
hello and thank you hensi, I managed to find a good connection, I now have the data display, I have changed the display colors, and you can enlarge the size of the information on the screen (with the screens editor)

[image]

14ri004, 2022-05-28:
I can't get the full width of the screen, it gives me the same size as the ili9225, so there's no point in installing a 9341, I have a white band on the right of the screen that doesn't bother me not authorize to use the entire surface in width.
is there a solution ??

[image]

dl9rdz, 2022-05-28:
This looks strange, this should be working fine.

Ah, maybe you have put something in your screens.txt too far to the right? The width of the display might not always be checked on all Display functions, and if you write something that goes beyond the rightmost border, strange things may happen.

dl9rdz, 2022-05-28:
Most people seem to use the large display in vertical mode (portrait). You could give it a try (TFT orientation 1 or 3), and use the existing screens5.txt. This is already a nice layout making use of the full display size.

14ri004, 2022-05-28:
thank you hensi, I am testing the different displays, I thought I had a larger display than on the ili9225 but the useful surface is identical between the 2, and the text is not larger either, it is an ili9341 in 2.8 Thumbs, shame not to be able to enjoy a 240x320 screen, THANK YOU

dl9rdz, 2022-05-28:
Well, as I said, it **should** work, you should be able to fully use the 240x320 screen area!

[image] [image]
(left: custom screens4.txt; right: standard screens5.txt included in the repository)

14ri004, 2022-05-28:
are 2.8 inch screens supported by TTGO??, vertical or horizontal I'm not the whole screen surface, this seems to be the file that only takes 2.0 inch screens, I I tested all possible positions but the result is identical, I cannot shift the text more than 12 to the right for it to still be visible, so I have a black band on the right of the screen no usable, which comes down to the size of the ili9225

dl9rdz, 2022-05-28:
No, this should not look like this. Usually ILI9341 work just fine.

Either you have some configuration error, or your display is somehow strange or broken.

Are you sure you have the display type set correctly (3, not 4)?

14ri004, 2022-05-28:
the screen is new, plugged into a TTGO v2.16, I took 3=ili9341, I tried all the screens from 1 to 5 and the result is identical, always this black band either on the right or in bottom (landscape/portrait), I don't see where the error would be????

[image]

dl9rdz, 2022-05-28:
Well, then I am running out of ideas. 

Maybe your display has an ILI9342 instead of an ILI9341 conroller? This is the last thing I can imaging. Try type 4 instead of type 3.

14ri004, 2022-05-28:
if I put the 4 the screen does not work, I do not know where it can come from, thank you for your help hensi

dl9rdz, 2022-05-28:
One more thing: Which version are you using?  Binary firmware image, or compiled the code yourself?

Maybe you can also try this simple example software from the display driver: https://github.com/moononournation/Arduino_GFX/blob/master/examples/HelloWorldGfxfont/HelloWorldGfxfont.ino
You need to adjust display type and pins in the Arduino code. 
Would be interesting to see if that works or shows the same issue.

14ri004, 2022-05-28:
wouldn't the problem come from the RGB screen instead of SPI????

14ri004, 2022-05-28:
is the screen compatible with the resolution of the TTGO v2.16??

[image]

dl9rdz, 2022-05-28:
Well, yes, it should be possible.

I just saw this from someone with a similar problem: https://forum.arduino.cc/t/adafruit-ili9341-not-using-the-whole-display/964672/21. In that case, it seems that the issue was that the display controller is not an ILI9341 but a ST7789. 

What I would suggest is to use the Arduino IDE, and some of the example applications from the Arduino_GFX library, and just test if these work correctly (with either ILI9341 or ST7789).

Or you can simply replace ILI9341 with ST7789 in Display.cpp if that is the problem

14ri004, 2022-05-28:
replace ili9341 by ST7789 in Display.cpp, it becomes complicated for me, I don't know how to make this modification and where to find this file

dl9rdz, 2022-05-28:
- Download this file: https://github.com/dl9rdz/rdz_ttgo_sonde/blob/gh-pages/devel/test-st7789-0x10000.bin
- Flash it to the ESP32 at address 0x10000 (!!!) (not 0x1000 like for the normal full .bin)
- On reboot, software on TTGO should show version devel7789test
- Set display type to 4 (usually this is ILI9342, but this test version uses the ST7789 driver in that setting)

14ri004, 2022-05-28:
Thanks, I'll give it a try and let you know how it goes.

14ri004, 2022-05-28:
Thanks hansi, works perfectly with the ST7789

[image]

14ri004, 2022-05-29:
Hello hansi, the file works perfectly with the screen, would it be possible to add it to the next Devel?, thank you for your help

dl9rdz, 2022-05-29:
Next devel should have a option 5 (ST7789) in the type of screen selection. (some code reorg, and not really well tested)

closed 2022-05-30 by 14ri004