dl9rdz/rdz_ttgo_sonde #304: Compile Error with Platform IO

smarti2019, 2022-05-29:
the last Messages from Compiling with Platform IO in Linux:

Checking size .pio/build/ttgo-lora32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (1310894 bytes) is greater than maximum allowed (1310720 bytes)
RAM:   [====      ]  37.4% (used 122532 bytes from 32*** [checkprogsize] Explicit exit, status 1
7680 bytes)
Flash: [==========]  100.0% (used 1310894 bytes from 1310720 bytes)
=============== [FAILED] Took 262.83 seconds ===============

dl9rdz, 2022-05-29:
Interesting.  Here it is that:
Flash: [==========]  100.0% (used 1310554 bytes from 1310720 bytes)
Assuming you use the most recent version. Maybe some minor library version difference on your system?

For now you can work around this issue by disabling some feature.
Either edit Display.cpp and for example comment out those two lines if you don´t not this display driver:
        else if(_type == 5) 
          tft = new Arduino_ST7789(bus, sonde.config.oled_rst); 
Or you can try disabling some feature in features.h (not tested if that still works)

smarti2019, 2022-05-29:
Hi, I'm currently using the last devel version. My Platform IO on Linux has just been completely freshly installed.

Disable some feature in feature.h is a working "Workaround". It compiled sussessfull  now.

hb9fxq, 2022-06-14:
Confirmed with PlatformIO Extension 2.4.3 on Linux (Ubuntu 20.04) 

`Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====   Error: The program size (1310926 bytes) is greater than maximum allowed (1310720 bytes)
   ]  37.4% (used 1225*** [checkprogsize] Explicit exit, status 1
32 bytes from 327680 bytes)
Flash: [==========]  100.0% (used 1310926 bytes from 1310720 bytes)`

Had to disable : 
`#define FEATURE_SONDEHUB 0
#define FEATURE_CHASEMAPPER 0`

thx @smarti2019 !

dl9rdz, 2024-09-04:
Closing this...
(Also note that the new dev version uses the platform espressif framework from the tasmota project, which has been optimized by that project to save space, so this should not be an issue any more. On the downside, you cannot easily compile with Arduino IDE any more (well you can, but this uses the standard Espressif version that results in a larger image, you have to select a different partitioning scheme, otherwise it will not fit)

closed 2024-09-04 by dl9rdz