dl9rdz/rdz_ttgo_sonde #308: Nearfield distance report

Pedritoprint, 2022-06-13:
Hello
During chasing my last three sondes, in nearfield, about 30m, the result of distance calculating shows abruptly and not with constant values a distance in minus meters. After a while, the Display jumps to normal distance with constant value and then again (-) with unstable values. How is it possible to calculate a minus distance? 
Is it possible that the signal in nearfield is to strong to demodulate the correct data? 
I have a **M5Stack** with rdz_ttgo software Vers. _**devel20220426-B16-full.bin**_  and an Moxon-Antenna
73 de HB9CMI/Peter

dl9rdz, 2022-06-13:
Hello Peter,
thanks for reporting this issue. I don't have a good explanation for that, this is not to be expected. The code is simply calulating d=sqrt(x*x+y*y), which should not yield negative values at all, and I never observed such behaviour myself, in spite of some testing I did with a test sonde just 10m to 20m away.
I don't thing signal strength is an issue. Either decoding fails completely, or if you get some frames decoded, they have correct CRC and thus should be valid.
Will do some more tests myself when I have some time available...

Pedritoprint, 2022-06-14:
Thanks Hansi for info and i will wait for next sonde nearby me to see if i could do some test or test carefully with an old sonde i have. 
It's not so urgent, because i found all sondes but it's strange, if the distance is going to minus and allways "jumping" values. Then i have to wait for a time till all is back to normal and then go on with hunting.

Pedritoprint, 2022-06-19:
Hansi, i think it's my mistake, because this week, if i recovered the sonde from Stuttgart (S1620174), i saw on the Display of the M5Stack, that the signalstrength is displayed above the valuable of distance. On daylight if the sun is mirrored, it's easy to mix up this both values, but i will do my further observation during next sondes.