dl9rdz/rdz_ttgo_sonde #309: ili9341 v1.2 screen problems in 2.8"

14ri004, 2022-06-14:
Hello Hansi, I have just received 2 news ili9341 v1.2 screens in 2.8", these are not the same screens as the previous version (ST7789), the 2 screens are well taken in Type-3 (ili9341), but also I find myself like the other day with a white band on the side of the screen, there is still text visible in this white band, it becomes really annoying, thank you in advance if there is a solution to this problem .

[image]

[image]

14ri004, 2022-06-15:
[image]

14ri004, 2022-06-18:
the 2.8" screen has a white stripe, while there is none on the 3.2"

the ili9341 2.8"
[image]

the ili9341 3.2"
[image]

dl9rdz, 2022-06-18:
Ok, strange, so the first question might be is this possibly a hardware defect?

Did you try using the display with different software? You could try example software from some of the (mulitple) available ILI9341 libraries. 
- https://github.com/moononournation/Arduino_GFX (the library I am using, it contains some example applications)
- https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9341-HX8347D
- https://github.com/loboris/ESP32_Adafruit_ILI9341
Note that they might require a different wiring or adjustment in the code...

14ri004, 2022-06-18:
Good evening Hansi, no I did not try the screens on anything else, I checked the wiring on the ttgo 3 times, the 2.8 screens are in version 1.2 and the 3.2 screen is in version 1.0, again our Chinese friends who changed something??

14ri004, 2022-06-19:
Hello hansi, the problem comes from a screen that would be out of service, I put the second screen which works perfectly, so problem solved, thank you

closed 2022-06-19 by 14ri004