dl9rdz/rdz_ttgo_sonde #311: help with arps for radiosondy.info

WaldperlachFabi, 2022-06-26:
Hello! Can someone help me to configure the ARPS for radiosondy.info? It doesn’t work me… What’s my mistake? Thank you for your help!
[image]

hr-ru, 2022-06-26:
You need to put your callsign in the Call entry 
You need to put in the correct APRS passvode in the passcode field
Call and APRS object call can be the same for radiosondy.info (they need to be same call, different SSID for wettersonde.net)

You don't need KISS TNC or AXUDP (but if you want to use them it you can enable them of course)
Everything else seems fime.

WaldperlachFabi, 2022-06-26:
Thank you! Where do i find my callsign and my passcode? Thank you!

hr-ru, 2022-06-26:
I guess from your question that you are not a radio amateur :) so no official callsign, which means for the purpose of radiosonde.info you can choose any nickname (WALDP2 as in your screenshot is fine).

You should also use port 14590 (non-forwarding) instead of 14580 (forwarding to arps-is), as the aprs-is is meant for amateur radio traffic only (and since arps.is is filtering out sonde data anyway, there is not much point in feeding data to aprs-is anyway). 

While the APRS passcode was originally meant to restrict traffic to radio amateurs only, it is a very week form of authentication, and google search will easily point you to sites that will generate a passcode for you.

TarmoKolli, 2022-06-27:
Go to radiosondi.info.
Log in with your username.
Open the user panel.
Click the "Rasperry pi system monitoring" button.
You'll see the name of your recipient and add a clip and sequence number.
For example: -11
Then you use the online APRS password generator to generate a password.

mmadela, 2022-11-13:
https://apps.magicbug.co.uk/passcode/
does the job
enter your login as callsign

closed 2022-12-31 by WaldperlachFabi