dl9rdz/rdz_ttgo_sonde #314: WhatsApp Notification Upon Sonde Detection?

ZigiWalter, 2022-07-17:
Hi,

I’m a big fan of the rdzTTGOSonde FW, so first – a big thanks to everyone involved!

One thing I’ve been missing is a push notification upon sonde detection which is - at times - more convenient than actively staring at the small screen or at the web GUI. To this end, I’ve tried doing a relatively small change in the FW to send a WhatsApp message that includes the ID and altitude of the a newly detected sonde, together with a link Google maps link.

I know that there is a big concern with the limited resources available on the ESP32 platform, so not all features can go in. Obviously, I will leave it for the maintainers of rdzTTGOSonde to decide whether this change (or a revised version) belongs in the main branch or not.
The notification is sent using a simple HTTP GET request to the free “CallMeBot” service API (https://www.callmebot.com/blog/free-api-whatsapp-messages), so the FW cost is quite low. Here are the required resources as reported during the compilation process:

Baseline (current head of “devel” branch”):
```
Sketch uses 1123762 bytes (85%) of program storage space. Maximum is 1310720 bytes.
Global variables use 118360 bytes (40%) of dynamic memory, leaving 176552 bytes for local variables. Maximum is 294912 bytes.
```
With the CallMeBot support:
```
Sketch uses 1125630 bytes (85%) of program storage space. Maximum is 1310720 bytes.
Global variables use 118488 bytes (40%) of dynamic memory, leaving 176424 bytes for local variables. Maximum is 294912 bytes.
```

You can review the changes in the “callmebot” branch here: https://github.com/ZigiWalter/rdz_ttgo_sonde/tree/callmebot
Initial testing looks good, but this definitely can use a better review.

Thanks.

LukePrior, 2022-07-17:
You could also look into hooking into one of the available data upload endpoints and use that to trigger the notification which would remove the need for more code on the device.

You could use MQTT, APRS, or SondeHub API along with a hosted script but that would require another service which might not be ideal.

ZigiWalter, 2022-07-17:
Thanks for the feedback!

I guess that would work too, but as you wrote it will probably require a script running in my "base station", opening a port for MQTT, and somehow working around the dynamically changing IP, right?
I do use the above approach for my home-based RPI running radiosonde_auto_rx, but when the receiver is not connected to my local WiFi, it becomes slightly less ideal.

LukePrior, 2022-07-17:
Yeah to be honest after having a look it doesn't really require much code so it seems like a good addition.

ZigiWalter, 2022-07-21:
@LukePrior - thanks!
@dl9rdz - should I go ahead and create a pull request?

ZigiWalter, 2023-04-01:
Pull Request: https://github.com/dl9rdz/rdz_ttgo_sonde/pull/354