dl9rdz/rdz_ttgo_sonde #320: elevation of the probe

14ri004, 2022-08-01:
Hello, is it possible to have the elevation of the probe on LiveMap compared to the GPS point of the TTGO, as on SondeHub

thank you

[image]

dl9rdz, 2022-08-06:
On the LiveMaop it should be no big issue, I can add it to the todo list (or will pull requests for that - I did not implement the LiveMap part myself).
On the TTGO, the calculation would require quite some part in the very small space for the code, so it is not likely to be added (or possibly configurable in source code, for compiling yourself if you leave away other parts to gain more free space)

14ri004, 2022-08-06:
hello, and thank you for your answer, too bad for the lack of space, this function would be interesting on the TTGO for monitoring Probes, thank you

closed 2022-08-06 by 14ri004