dl9rdz/rdz_ttgo_sonde #325: Can't configure at powerup with totally unexpected behavior

lawendel, 2022-09-18:
Instead of the expected first power-up info screen and with no trace of the RDZsonde SSID in my AP list, my newly flashed TTGO Lora32 enters into a weird, repeated 10 seconds scanning cycle and after a few iterations (each time the green LED blinks), the display goes black. What I did wrong?

[Here's a short YT video with what's going on on the board display](https://www.youtube.com/watch?v=5c6vuZ70jGM)

Apparently I successfully compiled and flashed under Arduino IDE 1.8 in MacOS High Sierra 10.13 the "devel20220531" file I downloaded from Github. All libraries were included, the .jar tool regularly unzipped in /Documents/Arduino/tools and so on. No compilation errors whatsoever from the IDE console, everything looks fine:

```
Lo sketch usa 1131786 byte (86%) dello spazio disponibile per i programmi. Il massimo è 1310720 byte.
Le variabili globali usano 118384 byte (40%) di memoria dinamica, lasciando altri 176528 byte liberi per le variabili locali. Il massimo è 294912 byte.
esptool.py v3.0-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d8:a0:1d:56:b1:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
```
I have repeated the operation, flashing a second time, but this behavior doesn't change, also if I press the reset button on the board and remove USB power. Do you suggest to repeat the compiling session with some other version of the source code? Should I try to bin-flash directly with ESPtools?

Thanks for your valuable software and support!
Andy

hr-ru, 2022-09-18:
You need to upload the file system part with all the config as well.

Either you flash the full .bin image (that contains everything).

Or if you prefer using ArduinoIDE, you need to follow these instructions:
https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Arduino-IDE-installation

Including the last line: "Upload data to SPIFFS with Tools -> ESP32 Sketch Data Upload"

lawendel, 2022-09-18:
Thanks for the prompt answer hr-ru, I followed that part of the wiki, but I can't find any "Tools -> ESP32 Sketch Data Upload" (or any similar command in the Italian version of Tools menu). Maybe it refers to Tools->Programmer-> at the bottom of Tools menu, but that says "no programmers available for this board". 
Could it be connected to the ESP32fs.jar file in ~/Documents/Arduino/tools/ESP32fs/ ?

hr-ru, 2022-09-18:
Yes, you need to install the ESP32fs extension (ESP32fs.jar) and restart ArduinoIDE.
Then you should have this Menu item available.

lawendel, 2022-09-18:
Yes, i did that, the .jar file is there, I restarted the IDE, but that particular menu item is still missing. Maybe it should not be located in /Documents/Arduino/tools I need to check. Thanks a lot, now I know what went wrong.

lawendel, 2022-09-18:
Solved. Don't ask why, ArduinoIDE Tools menu now includes an ESP Sketch Data Upload which brilliantly flashed the compiled code. Initially I had simply quit the program and restart after uploading ESP32fs.jar in /Documents/Arduino/tools, now I closed the sketch windows, quit, restart, loaded the source with .ino and recompile. Boom. Maybe it was trivial for everyone, not for me! Anyway, issue closed, thanks for pointing me at the right direction. If I may suggest a slight mod to the Arduino IDE section of the wiki, where ESP32fs is mentioned: 

> Unzip the content to the tools folder of your Arduino IDE (~/Documents/Arduino/tools on MacOS, similar on other OS)

You might specify that On MacOS, if /Documents/Arduino/tools doesn't exist, one simply has to create it. At some point I got confused when I found another tools folder in ~/Library/Arduino15/packages/esp32/tools.

Thanks again!

dl9rdz, 2022-10-01:
Ok thanks, I have added a sentence to the documentation regarding the creation of the folder.

closed 2022-10-01 by dl9rdz