dl9rdz/rdz_ttgo_sonde #331: probe distance in Meters

14ri004, 2022-11-02:
good evening, when the probe is more than 100km away it's good, but at 99km the distance indicates meters, would it be possible to have the distance in meters from only 1000 meters, thank you

[image]

hr-ru, 2022-11-02:
In the screens file, you can change GD to GD5 for getting a 5-digit-version (including the unit), so it will be km for anything >= 10km, and meters below (i.e. you will have 10.0km but then 9999m). Is that good enough?

14ri004, 2022-11-03:
Hello, I tried GD5 but unfortunately it does not work, I always have the figures in Meters below 100km

14ri004, 2022-11-25:
hello, still no solution to have the distance of the probe in Meters and only below 10km, up to 100 km it's ok, but below it goes to 99999 m instead of 99 km, I have test the GD5 manipulation but it does not change the result, is there a solution, thank you

dl9rdz, 2022-11-25:
Are you sure you changed the right file and/or screen? I cannot test it myself right now, but this *should* work.
I can check myself later some time in December.

14ri004, 2022-11-25:
hello, yes change made in the correct screen and on 2 ttgo, it does not change the result on the distance of the probe, above 100 km ok, but below it is in meters 99999m

kmp56, 2022-12-17:
confirmation,  GD5 change nothing....  (LoRa32 V2.1 _ 1.6)

dl9rdz, 2022-12-17:
Alright, there is a bug. Or was. Should be fixed in devel20221217 (without testing)

kmp56, 2022-12-18:
≤ 100km is ok, still km, e.g. 42km also km. I can't check ≤ 10km - the probe is too far unfortunately. I hope there will be meters...

dl9rdz, 2022-12-18:
Yes, <10km it should be meters.

14ri004, 2022-12-18:
Hello, indeed it works in KM below 100 km, I'm waiting to see below 10km, but it seems perfect to me, thank you and have a good Sunday

kmp56, 2022-12-18:
So everything should work ok ;) BTW the test sonde on my balcony shows correctly - meters. 
Thanks for the great job!  Merry Christmas!

14ri004, 2022-12-18:
[image]

14ri004, 2022-12-18:
[image]

closed 2022-12-19 by 14ri004