dl9rdz/rdz_ttgo_sonde #334: Question regarding the QRG

lelun, 2022-11-18:
Hi i have just reviced my ttgo and managed to flash it, i normaly use a RTL-SDR for ballon opservations but i hope the ttgo will automate ballon opservations for me and allow me to better judge if i want to go on a hike or not.

will the ttgo look at other frequencys then the ones defined in the QRG? 
my local launch site have a habbit of randomly changing frequencys and there is no good way to make a QRG for this station.
so during todays test run i manualy defined the frequency after checking with my rtl-sdr, is it able to look for a strong signal on its own and start decoding on its own without a defined QRG entry?

also there is the plaucebility of 2 types of sondes operating in my area will it be able to switch from the M20 / RS41 on its own with out a QRG or do i have to set up 2 sets of QRG for both sonde types?

i was unable to find answers to thise questions on the wiki, but i have found simular topics discussed before.

otherwise i want to say i really like this software and it looks like it. it was really relieble during todays test run after i connected it to my rtl/sdr antenna and manualy defined the freq via the QRF. [todays opservation](https://sondehub.org/208-2-00292)

device is a TTGO LORA v2.1_1.6  software version is rdzTTGOsonde-master_v0.9.2

dl9rdz, 2022-11-18:
Hi,

The ttgo will not look at other frequencies. Unlike the RTL-SDR, which can scan a 2 MHz wide band simultaneously,
the TTGO is limited to receive a single frequency. You can define a list of frequencies it iterates over. It also cannot
decode multiple sonde types in parallel, each type needs a separate entry in the list (except for very similar decoders,
like DFM6/DFM9, or M10/M20, these can select the exact type automatically)

There is one feature which allows automated updates of frequencies over the Internet using sondehub
(or other services with same API). This does not enable the detection of completely new signals, but if someone else
is receiving a signal at a location close to you and uploading that to sondehub. the corresponding frequency can
automatically be added to your list on the TTGO.

Cheers, Hansi, dl9rdz

lelun, 2022-11-19:
hi thanks for the responce.
how wide of a band can it scan around the frequency i have set? 
i am hoping i can set a tight enugh QRG so the error correction can help with getting a solid signal, 

other wise it looks like there is a recvicer station nearby that is only able to pick up the sonde for a few hundred meters after launch and post the frequency so i will try set mine up to send frequency requests and hopefully i can use that as a backup.

also if you dont mind me asking i have a feature request. i would love to have a way to shedual frequency request form sondehub that is not based on time intervals but on utc. so i know when sondes are launched from the station i am looking at and i would like for my ttgo to only send 3 frequency requests with a 10 min interval at a designated time. this is to reduce spam for sondehub from my ttgo.

and can you add a smal note the the set up instructions that if you cant find the ttgo on your comports to check if the usb cable actually has data lines and is not just for charging. i made that mistake during my settup and i just want to note that as a plauceble issue for those that are inexperienced 

other then that agian i have to say i really like this software and think you have done a really good job with both the insturctions and wiki. my compliments!
good day :)

closed 2022-11-19 by lelun

dl9rdz, 2023-06-25:
paid

> Begin forwarded message:
> 
> From: ***@***.***>
> Subject: [CS_ANACOM-S015460/2022] Dados de pagamento - alteração do documento CAN/licença CEPT
> Date: 31 October 2022 at 10:52:59 GMT
> To: ***@***.***>
> 
> 000315654