dl9rdz/rdz_ttgo_sonde #337: Location of tracker not on Sondehub

efiten, 2022-11-30:
**Describe the bug**
When I run my tracker (ESP32 or T-beam), the location is not published in the Sondehub map.

**To Reproduce**
This happens to both trackers that I test, always.

**Expected behavior**
Location visible

**Version and specific configuration:**
 - What firmware version are you using? Used master BIN file (version master_v0.9.2)
 - Hardware you are using? ESP32 board & TTGO T-beam

**Additional context**
I've set the mode to FIXED and filled in my GPS location :
[image]
On the livemap inside the internal page, I'm visible : 
[image]

efiten, 2022-12-01:
Left the device online for some, and it looks like it doesn't even sent the data to Sondehub. Do I miss something? Setting reporting to '1', set the callsign, that's it ?

14ri004, 2022-12-13:
[image]

hello, why Rdzsonde-2 ????

efiten, 2022-12-13:
Because I'm running 2 board, as a test, so I can see what board I'm working
with during test.

titoaprendiz, 2022-12-24:
Olá!
Eu também estava com o mesmo problema, mas tudo foi resolvido com a instalação de novo browser.
Eu usava chrome e fireFox, e dava erro nos dois navegadores
Fiz a instalação do Opera, mas não importei as configurações dos dois navegadores.
Refiz todo procedimento de instação e tudo funcionou perfeitamente bem.
73 de Tito Augusto, PU2OZK.

DG0MG, 2022-12-27:
@efiten 
Please try to set "SondeHub location reporting" to **3** (automatic, not fixed). 
I remember, i had the same problem at the beginning of my fixed receiver station.

14ri004, 2022-12-27:
Merci pour votre réponse, je vais essayer cette solution et voir si cela fonctionne, merci Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : DG0MG ***@***.***> Date : 27/12/2022  16:12  (GMT+01:00) À : dl9rdz/rdz_ttgo_sonde ***@***.***> Cc : 14ri004 ***@***.***>, Comment ***@***.***> Objet : Re: [dl9rdz/rdz_ttgo_sonde] Location of tracker not on Sondehub
  (Issue #337) 
@efiten
Please try to set "SondeHub location reporting" to 3 (automatic, not fixed).
I remember, i had the same problem at the beginning of my fixed receiver station.

closed 2024-09-04 by dl9rdz