dl9rdz/rdz_ttgo_sonde #338: Support for SX1268

mph88, 2022-12-11:
**Describe the bug**
The SX1268 does not turn on. The serial output shows that communication with the chip does not occur, as the frequency setting nor gain settings at startup apply. Additionally it reports power on FAILED and the frequency plot shows a flat line. Communication of the ESP32 over wifi, configuring settings, GPS all work ok and unit does not panic and restart during failure to connect to SX1268.

**To Reproduce**
Flashed with PlatformIO using latest dev version.

**Expected behavior**
Chip turns on and is tunable.

**Screenshots**
N/A

**Version and specific configuration**:
Board: TTGO T-Beam V1.1 SX1268
version flashed: devel20221113

**Additional context**
Serial Output:
`Configuing SX1278FSK SPI with miso=19, mosi=27, sck=5, ss=18`
`170`
`Setting freq: SUCCESS `
`Frequency set to 682666624.00`
`170`
`Setting freq: SUCCESS `
`Frequency set to 682666624.00`
`RX LNA Gain is -36`
`SX1278 configuration finished`
`Setup finished`
`...`
`MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 125328, unused stack: 5208`
`170`
`Setting SX1278 power on FAILED`
`Sonde::setup() done: Type RS41 Freq 682.666624, AFC BW: 100000, RX BW: 100000`
`TOO MUCH DATA`
`Sonde::timeoutEvent: NORX`
`Sonde:receive(): result 1 (RX_TIMEOUT), event 0b => action 93`
`RXtask: start DECODER for sonde 19 (was DECODER)`
`Sonde::setup() start on index 19`
`waitRXcomplete returning 9301 (RX_TIMEOUT)`
`MAIN: loopDecoder: action 93 (Sonde=19) => 65  [current: main=19, rxtask=19]`
`MAIN: updateDisplay started`
`MAIN: updateDisplay done (after 71 ms)`

dl9rdz, 2022-12-11:
Thanks for the report. Yes, the SX1268 is a different chip, currently only the SX1278 is supported. 
The chips are different, at least frequency programming for SX1268 is different, so I'll have to see if this can be added to the firmware. But unfortunately I don't have any SX1268 for testing. If I manage to get one delivered to Germany before Jan 1st I might give it a try...

au51gtm, 2023-02-22:
Any progress on the SX1268 support?

dl9rdz, 2023-02-22:
Unfortunately not, still haven't got a board to test myself.

au51gtm, 2023-02-22:
I can provide a board on loan if that helps.  I can't use it until the 
firmware is updated, in any case.

miniarci, 2023-12-07:
Hi! @dl9rdz 
Any news in topic implementation SX1268
If you need someone to test i can help :)