dl9rdz/rdz_ttgo_sonde #340: Télémetry Data

14ri004, 2022-12-27:
Hello, on screen 14 of the RDZ, there is the Telemetry Data, but even during the decoding of a probe the atmospheric pressure is not displayed on the screen (MphPa), is there a solution for have it visible on the T-BEAM, thank you

dl9rdz, 2022-12-27:
The only sonde for which that is supported is RS41-SGP. 

If its a RS41-SG (without pressure sensor) or any other type, pressure will not be displayed.

If you are receiving a RS41-SGP, it will take a while because before decoding pressure, the relevant calibration data needs to be received. This can take around one minute or so (assuming all frames are received error-free in that period).

14ri004, 2022-12-27:
OK merci

closed 2022-12-27 by 14ri004