dl9rdz/rdz_ttgo_sonde #341: Radiosondy does not receive the last balloon reading

rudsomlima, 2022-12-30:
Radiosondy does not receive the last balloon reading, SondeHub does.
version master_v0.9.2.

Example:

https://radiosondy.info/sonde_archive.php?sondenumber=U0440587
[image]

https://sondehub.org/#!mt=Mapnik&mz=11&qm=12h&mc=-6.11325,-35.21358&f=U0440587
[image]

Thanks.

dl9rdz, 2022-12-30:
Yes, this is a known limitation. Mainly due to the fact that radiosondy gets all data (position update every second), but due to constraints of aprs, in the default configuration only one position update is sent every 10 seconds. 

You will see the same effect with most other software as well.

rudsomlima, 2022-12-30:
> 
Could you wait about 60s and then upload to radiosondy the last received data?

I like radiosondy because it is able to show all probe landings by area:
https://radiosondy.info/maps/web_map.php?kml=1&search_limit=100&startplace=Natal%20(BR)

dl9rdz, 2022-12-30:
I known one software is doing that, but given the nature of APRS and how it is used by radiosondy (timestamps by when a packet arrives at the server, not from the decoded message), this is (in my opinion) a very annoying feature: If someone else is receiving the sonde for a bit longer (lets say 10 or 20 seconds), and then you send your last position 60 seconds later, then radiosondy will show that delayed transmisssion as "last" position, not the real last position of someone else.