dl9rdz/rdz_ttgo_sonde #343: FTP Ephemeris Rinex

RegsJR, 2023-01-09:
Hi, all,
there seems to be an error while trying to download Ephemeris data from FTP servers. Last year was the FTP replaced with HTTPS protocol so NASA or EASA data are no longer available via FTP. I tried to download from Bund.de like shown in the code below, but without success. How to solve this, please?

```
FTP connection to igs-ftp.bkg.bund.de failed
Skipping header: 
reading RNX done, result is 0, ephs=0x0
SetupAsyncServer()

sondehub_station_update()
```

hr-ru, 2023-01-09:
The server in the current default configuration should still be available

ephftp=gssc.esa.int/gnss/data/daily/

```
hans@s1:~$ ftp gssc.esa.int
Connected to gssc.esa.int.
220 Wing FTP Server ready...
Name (gssc.esa.int:hans): anonymous
331 Password required for anonymous
Password:
230 User anonymous logged in.
Remote system type is UNIX.
Using binary mode to transfer files.
ftp> cd /gnss/data/daily/2023/009
250 CWD command successful. "/gnss/data/daily/2023/009" is current directory.
ftp> ls
200 Port command successful
150 Opening data channel for directory list.
-rw-rw-r-- 1 5510 gssc_adm          43955  Jan 09 16:56 brdc0090.23n.gz
...
```

RegsJR, 2023-01-09:
Still the same status, here
...
running geteph

FTP connection to gssc.esa.int failed
Skipping header: 
reading RNX done, result is 0, ephs=0x0
SetupAsyncServer()
...

dl9rdz, 2024-08-16:
The ftp server is still working for me, so closing this issue for now. Maybe in the long term we might switch to http[s], but for now ftp download should work [unless you are on a very restrictive network that actually blocks ftp]
ephftp=gssc.esa.int/gnss/data/daily/%04d/brdc/brdc%03d0.%02dn.gz

closed 2024-08-16 by dl9rdz