dl9rdz/rdz_ttgo_sonde #344: rdzTTGOsonde v0.9.2 seems to decode DFM17 as DFM09P

AstroTaka, 2023-01-14:
I am a user of auto_rx.
I received DFM17 sonde signal.
My station could not upload a telemetry to SondeHub because another rdzTTGOsonde station already upload a telemetry as DFM09P.
https://sondehub.org/21063210

Currently, Japan government only approve to use DFM-06 and DFM-17 of Graw sonde.
So nobody can use DFM09P in Japan.

(This is a JAPANESE site to search approved sonde of "DFM-xx")
https://www.tele.soumu.go.jp/giteki/SearchServlet?pageID=jk01&NUM=&NAM=&FOM=DFM&PC=&YAR_FROM=&MON_FROM=&DAY_FROM=&YAR_TO=&MON_TO=&DAY_TO=&RAD=02-03-02-00&TEC=1&TEC=2&TEC=3&TEC=4&TEC=5&TEC=6&TEC=7&SK=0&DC=0&SC=1&as_fid=c22f7ab117dd061276103c024124329680157d3b#searchlist

And rdzTTGOsonde v0.9.2 decode DFM17 as DFM09P I think.

Please check.

I received the previous DFM17 sonde.
https://sondehub.org/21063324
The both ID is similar.

AstroTaka, 2023-01-14:
I checked again about upload telemetry.
I have no issue about upload a telemetry of the same sonde of "DFM17" and "DFM09P" into SondeHub.

I compared the both telemetry.
DFM17 decoded telemetry has "Temperature External:".

https://sondehub.org/21065517

dl9rdz, 2023-08-21:
I assume this indeed has been fixed, closing the issue...

closed 2023-08-21 by dl9rdz