dl9rdz/rdz_ttgo_sonde #348: Frequency exclusion.

Szetya, 2023-02-03:
Frequency exclusion.
How can I mark a frequency in the QRG list so that the SondeHub frequency import does not reactivate it?
I want to use the frequency importer but don't want the receiver to "hang up" on a nearby radiosonde.
Thanks!

lelun, 2023-02-19:
do you have a standby frequency that is sett on the qrg?

i have 2 TTGO devices deployed 1 is acting as a master and other as slave 
the master one has a almost full qrg that has all of the common freqencys that sondes are on, wile the other one is a suport device that has only 1 frequency manualy inputtet but it recives the frequency from the other TTGO.

if you have one that has a full qrg you could either make the sonde hub frequency import radius smaller, or lower the max time you are importing frequencys for. for me the frequencies that are older then 48 hours get automaticly removed. 

but yeah having a manualy sett standby frequency is the way i operate it and it works well.

Szetya, 2023-02-25:
I gave it a try with the following changes. It seems to work.

https://github.com/dl9rdz/rdz_ttgo_sonde/blob/devel/RX_FSK/src/ShFreqImport.cpp#L77
- sonde.sondeList[i].active = 1;
+ if(sonde.config.sondehub.fireactivate >= 1) sonde.sondeList[i].active = 1;

https://github.com/dl9rdz/rdz_ttgo_sonde/blob/devel/RX_FSK/data/cfg.js#L76
+ [ "sondehub.fireactivate", "Re-activate inactive sonde (0=no,1=yes)" ],

https://github.com/dl9rdz/rdz_ttgo_sonde/blob/devel/RX_FSK/src/Sonde.h#L226
+ int fireactivate;

https://github.com/dl9rdz/rdz_ttgo_sonde/blob/devel/RX_FSK/RX_FSK.ino#L726
+ {"sondehub.fireactivate", 0, &sonde.config.sondehub.fireactivate},

closed 2023-03-31 by Szetya