dl9rdz/rdz_ttgo_sonde #355: Imperial Units

Ernie34, 2023-04-05:
Is there a way to give the option to select metric or imperial units in the setup?

Thank you!!!

Ernie

dl9rdz, 2023-04-17:
Currently there is no such option, but I could put that on the TODO list (with low priority).

I would not change any units in network APIs, there it is up for the remote software to convert things according to user preferences,

For the LCD or TFT display, what would be the right units for those values:

- horizontal speed: currently configurable km/h or m/s => mph or ft/s?
- vertical speed: currently m/s => ft/s?
- altitude: current m => ft?
- distance: this is more complicated due to the limited space on the small LCD display.
currently it is either 5 digit (automatic: either "999km" or "9999m" if below 10km), or 6 digit (9999km or 99999m if below 100km) => 5 digit (999mi / 9.9mi if below 10mi / 999ft if below 1000ft)? 6 digits 9999mi..9.99mi / 9999ft if below 10000 ft?  Sorry, I never use imperial units, so no idea what is usually used.

wrevenaugh, 2023-12-02:
Hans, it would be **ABSOLUTELY WONDERFUL** if you could output imperial!
Horizontal Speed - Miles Per hour
Vertical Speed - Miles per hour
Altitude - Feet
Distance: Miles, then below 1 mile - Feet.
THNAK YOU SO MUCH!
Bill

bazjo, 2023-12-02:
The imperial option for vertical speed that our software supports / our imperial customers use is ft/s.
Also, maybe decimal feet below 1 mile would be an easier option?

flyincajun, 2024-05-07:
Could the local machine livemap.js feed be modified to show imperial units, or is that data part of the API?

dl9rdz, 2024-05-07:
The API delivers metric values, and I would not change that.
It would be feasible to add conversion code to livemap.js.

flyincajun, 2024-05-08:
Cool. Just want the live map to be imperial. I'll poke around and see how bad I mess it up.  😀

wrevenaugh, 2024-05-08:
To be clear guys, I'm looking to get Imperial on the TTGO display itself.  @flyincajun - would you have any idea where in the RX_FSK.ino (and accompanying sub directories) to go to accomplish this?  Thanks, Bill

dl9rdz, 2024-05-08:
For the TTGO display, this needs to go into Display.cpp. I have on my TODO list, I guess I will have time for that around June this...

wrevenaugh, 2024-05-08:
@dl9rdz Thanks!  I'll check that out.  Can you also please help me to find people or some help (or where to go to look) who have used bigger screens ?  I have not had much luck  there.  I have gotten it to display on some, but it's always disproportionate, and the fields and values are all screwed up.