dl9rdz/rdz_ttgo_sonde #357: Support for ESP32-S3

j0uni, 2023-04-11:
Could we have support for ESP32-S3. Heltec Lora V3 is using -S3 version of ESP32.

dl9rdz, 2023-04-11:
The -S3 itself should not be a big issue. Not sure if it is binary compatible, but just recompiling the source for that target should be good enough.

However, the main issue with the Heltec Lora V3 is that it has a newer radio chip (SX1262/SX1268), which is quite different from the SX1278 in the other boards.  What this means is that
- a lot of software adjustments required. The API of the chip is quite different.
- unsure if it works at all. 400 MHz is outside the regular specs of the chip (same for SX1278), but it needs to be tested to see if (or how well) it works on that frequency.
See also #388

5B4APV, 2023-04-21:
I came here to post the exact same request. I just received a few Heltec Wireless Sticks V3 and I couldn't get them working. It would be nice to have support as they seem nicely build and cheap boards.

dl9rdz, 2023-04-21:
I have now received a board with a SX1268, so there is a first step achieved :-). And I got some info from Semtech support on how to use it with an unlimited length packet format (more tricky and not in the data sheet, but seems feasible). 

So the next step is to check if it works on 400 MHz (spec is 410-810 MHz). It may or may not work...

j0uni, 2023-04-21:
> I have now received a board with a SX1268, so there is a first step achieved :-). And I got some info from Semtech support on how to use it with an unlimited length packet format (more tricky and [...]

So cool! If you need any help testing on Heltec V3, just let me know!

Illenganavi, 2023-07-17:
G'Day everybody,
is there any progress to support the SX1268? Thanks heaps!! :

LW6DSM, 2023-08-28:
Hello i have a Heltec ESP32 V3 any progress today Thanks!