dl9rdz/rdz_ttgo_sonde #361: DFM-17 temp decode differences between rdz_ttgo_sonde and radiosonde_auto_rx

markdconner, 2023-04-12:
I occasionally run both radiosonde_auto_rx and rdz_ttgo_sonde at my home near the US National Weather Service office in Valley, Nebraska to track the DFM-17 sondes launched from there.  I've noticed differences in the decoded temperatures between the two software packages.  An example is attached from last night's sonde (S/N 23002287, launched 11 Apr 2023 2300 UTC), which graphs the temperature decode results as archived at sondehub.org.  I've also attached the radiosonde_auto_rx log file in case that's helpful - I'm not sure how to obtain the equivalent for the rdz_ttgo_sonde data.  Also included is the official sounding obtained from http://weather.uwyo.edu/cgi-bin/sounding?region=naconf&TYPE=TEXT%3ALIST&YEAR=2023&MONTH=04&FROM=1200&TO=1200&STNM=72558

![2023-04-12 10_57_22-Compare Software - Grafana](https://user-images.githubusercontent.com/49458752/231514687-953bd369-3765-45aa-903b-0ece3a2ae5f
[OAX sounding 12 Apr 2023 00Z.txt](https://github.com/dl9rdz/rdz_ttgo_sonde/files/11213577/OAX.sounding.12.Apr.2023.00Z.txt)
3.jpg)
[20230411-222827_DFM-23002287_DFM17_403209_sonde.log](https://github.com/dl9rdz/rdz_ttgo_sonde/files/11213538/20230411-222827_DFM-23002287_DFM17_403209_sonde.log)

dl9rdz, 2023-04-12:
Basically I was using exactly the same code as auto_rx for the DFM temp calculation, but it indeed there was a later change in auto_rx:
-    if (gpx->ptu_out >= 0xC) {
+    if (gpx->sensortyp0xC == 'P') {  // 0xC: "P+" DFM-09P , "T-" DFM-17TU ; 0xD: "P-" DFM-17P ?
This is likely the cause for the discrepancy...

dl9rdz, 2023-04-12:
Actually, this likely is the same issue as #344.
My code is assuming that 0x0C is a DFM-9P. Actually it might be either a DFM-09P or a DFM-17. I put this on my TODO list...

dl9rdz, 2023-04-13:
Please check devel-20230413 if that fixes the problem.

markdconner, 2023-04-13:
I have updated to this version and will have it running for the 14/00Z launch

markdconner, 2023-04-14:
Temperature data matches perfectly now
[image]

dl9rdz, 2023-04-14:
Great, then I can close this issue...

closed 2023-04-14 by dl9rdz

darksidelemm, 2023-06-21:
Will there be another master release anytime soon with this update? 

There's a lot of stations running old versions of rdz_ttgo_sonde, resulting in temperature oddities...

hr-ru, 2023-06-21:
Hi, yes, I guess I should just release that. I wanted to wait to get a confirmation if #344 is also fixed by that, but I guess I can just assume that it is.

darksidelemm, 2023-06-21:
I think it's working. In this example we have a station running both auto_rx 1.6.1, and rdz 20230427:
https://grafana.v2.sondehub.org/d/dtBeyfyVk/compare-radiosonde-callsign-telemetry?orgId=1&var-Serial=23016345&from=1687346129912&to=1687348605931

The temperature data matches!