dl9rdz/rdz_ttgo_sonde #363: Building with PlatformIO and binary output is the wrong size

gariac, 2023-05-14:
First let me admit I am totally new at this.  I want to learn how to compile the binary then look at the code to learn. Kind of like taking a watch apart.  I can compile with PlatformIO  but the file size doesn't match the downloadable binary.

Processing ttgo-lora32 (platform: https://github.com/platformio/platform-espressif32.git#v3.3.2; board: ttgo-lora32-v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v1.html
PLATFORM: Espressif 32 (3.3.2+sha.e170b13) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting RX_FSK.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.34.17
|-- AXP202X_Library @ 1.1.3
|-- MicroNMEA @ 2.0.6
|-- ESP Async WebServer @ 1.2.3
|-- GFX Library for Arduino @ 1.1.5+sha.38a28fa
|-- AsyncMqttClient @ 0.8.2+sha.7eacb02
|-- ESPmDNS @ 1.0
|-- SPI @ 1.0
|-- SPIFFS @ 1.0
|-- Ticker @ 1.1
|-- Update @ 1.0
|-- WiFi @ 1.0
|-- Wire @ 1.0.1
Building in release mode

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'scripts/makefontpartition.py'
File "/home/mrg/.platformio/penv/lib64/python3.6/site-packages/platformio/builder/main.py", line 181, in <module>
Compiling .pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/Chasemapper.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/DFM.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/DecoderBase.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/DefaultFonts.c.o
Compiling .pio/build/ttgo-lora32/src/src/Display.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/M10M20.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/MP3H.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/RS41.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/RS92.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/SX1278FSK.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/Scanner.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/ShFreqImport.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/Sonde.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/TFT22_ILI9225.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/aprs.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/fonts/fonts.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/geteph.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/json.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/mqtt.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/nav_gps_vel.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/rs92gps.cpp.o
RX_FSK/src/mqtt.cpp: In member function 'void MQTT::publishPacket(SondeInfo*)':
RX_FSK/src/mqtt.cpp:70:16: warning: unused variable 's' [-Wunused-variable]
     SondeData *s = &(si->d);
                ^
Compiling .pio/build/ttgo-lora32/src/src/rsc.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/rsc_decode.cpp.o
Generating partitions .pio/build/ttgo-lora32/partitions.bin
Compiling .pio/build/ttgo-lora32/lib325/SPI/SPI.cpp.o
RX_FSK/src/Display.cpp: In member function 'virtual void U8x8Display::drawQS(uint16_t, uint16_t, uint8_t, uint8_t, uint8_t*, uint16_t, uint16_t)':
RX_FSK/src/Display.cpp:379:32: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[... 288 lines cut ...]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/home/mrg/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pio/build/ttgo-lora32/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/main.cpp.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/ttgo-lora32/libFrameworkArduino.a
Indexing .pio/build/ttgo-lora32/libFrameworkArduino.a
Linking .pio/build/ttgo-lora32/firmware.elf
Retrieving maximum program size .pio/build/ttgo-lora32/firmware.elf
Checking size .pio/build/ttgo-lora32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  37.4% (used 122492 bytes from 327680 bytes)
Flash: [==========]  100.0% (used 1310358 bytes from 1310720 bytes)
Building .pio/build/ttgo-lora32/firmware.bin
esptool.py v3.1
Merged 1 ELF section

hr-ru, 2023-05-14:
Not sure what exactly you are referring two, but there are two things you might want to consider:

(1) The pre-built binaries on http://rdzsonde.mooo.com/ are built with the Arduino IDE, not with PlatformIO. There are some minor differences in the runtime environment of both, so if you generate the binary with pio, it will not exactly match the one you generate with Arduino IDE.

However, in any case, both versions should be fully functional, so no difference in that regard.

(2) The "full" images on http://rdzsonde.mooo.com/ contain an (almost) complete flash memory image. These you can write (starting at offset 0x1000) to the flash and they have size 4 Mib (well, minus the 0x1000 offset to be fully accurate).
These full images contain (a) the program image, (b) space for the OTA update of the program image and (c) the file system image

The binary you create above with PlatformIO is just the program image. To generate (and upload) the file system image with PlatformIO, you can use pio run --target=uploadfs

hr-ru, 2023-05-14:
To be more accurate:
The images xxx.full.bin contain a full flash image starting at offset 0x1000. This includes at various offsets:
0x1000: bootloader
0x8000: partition table for the flash
0xE000: bootapp0
0x10000: the actual firmware (that is your .pio/build/ttgo-lora32/firmware.bin)
0x150000:  (space for the firmware to be written by OTA)
0x290000: the spi file system (.pio/build/ttgo-lora32/spiffs.bin generated by pio run --target=buildfs)

If you want to generate a "full" image with platformIO, you could use the script in scripts/makeimage.py, which creates that the full image using the individual parts (but based on the Arduino build, so you might have to adjust it a bit).

If you just want to write the firmware to a board using PlatformIO, you could simply use
pio run --target=uploadfs
pio run --target=upload
The first command writes the file system partition, the second command all other partitions.

gariac, 2023-05-15:
I noticed one of the "issues" mentioned PlatformIO so I assumed that was how the code was built.  

So  I see there are lines at the bottom of the ini file to uncomment.
`; Add / remove the following two lines for separate fonts partition in flash
; after changes:
; - pio run --target=upload (uploads partition table)
; - pio run --target=uploadfs (uploads file system matching the new partition table)
; if enable also do:
; - pio run --target=uploadfonts (uploads fonts.bin created during the first pio run)
; Then everything should be "back to normal"
;
extra_scripts = post:scripts/makefontpartition.py
;board_build.partitions = partition-fonts.csv
`
I gather a "normal" compilation would uncomment these lines:
` - pio run --target=upload (uploads partition table)
 - pio run --target=uploadfs (uploads file system matching the new partition table)
`
So what do I uncomment for the first run? Or could I just build the fonts every time and waste some CPU cycles.

Incidentally here are the names of the files being created. I find it old that that exist in a hidden .pio directory.
`
-rw-r--r-- 1 mrg users  1310464 May 14 14:28 firmware.bin
-rwxr-xr-x 1 mrg users 17999668 May 14 14:28 firmware.elf
drwxr-xr-x 3 mrg users     4096 May 14 14:28 FrameworkArduino
-rw-r--r-- 1 mrg users    15774 May 14 22:17 idedata.json
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib123
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib207
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib2e7
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib325
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib47a
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib51b
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib62c
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib793
drwxr-xr-x 3 mrg users     4096 May 14 14:27 lib7e7
drwxr-xr-x 3 mrg users     4096 May 14 14:27 liba1c
drwxr-xr-x 3 mrg users     4096 May 14 14:27 liba24
drwxr-xr-x 3 mrg users     4096 May 14 14:27 liba63
drwxr-xr-x 3 mrg users     4096 May 14 14:27 libc27
drwxr-xr-x 3 mrg users     4096 May 14 14:27 libc38
drwxr-xr-x 3 mrg users     4096 May 14 14:27 libfe5
-rw-r--r-- 1 mrg users 25129022 May 14 14:28 libFrameworkArduino.a
-rw-r--r-- 1 mrg users        8 May 14 14:27 libFrameworkArduinoVariant.a
-rw-r--r-- 1 mrg users     3072 May 14 14:27 partitions.bin
drwxr-xr-x 3 mrg users     4096 May 14 14:27 src

The top of the tree is 
~/Documents/PlatformIO/Projects/ttgo/.pio/build/ttgo-lora32
`

hr-ru, 2023-05-15:
Well, don´t overcomplicate. Usually the only thing you need to do is

pio run --target=uploadfs
pio run --target=upload

You need to do this only once, and no other changes needed. If you do not change any files in the file system, but only change code, you then can do the upload without re-uploading the file system (advantage: you keep all your configuration).

The variant with a separate font partition is not used by the normal build, so best ignore it :) Do not uncomment unless you know exactly what are you doing and why. (It would make more space available in the program partition. There was (or likely still is) a bug in the ArduinoIDE that makes it difficult to use a custom partition there. Mainly for this reason I am not using that variant right now)

gariac, 2023-05-15:
This all sounds good. I thought I needed the fonts. Thanks.

Less is more!

gariac, 2023-05-15:
Oops too soon. I get these file creation error. I know I can fix the problem by creating the directories, but is there some file permission trick I am missing?
`
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/src/Chasemapper.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/src/Chasemapper.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/src/DecoderBase.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/src/DecoderBase.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/src/DFM.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/src/DFM.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/src/MP3H.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/src/MP3H.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/src/RS41.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/src/RS41.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/src/M10M20.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/src/M10M20.cpp.o] Error 1
RX_FSK/src/Display.cpp: In member function 'virtual void U8x8Display::drawQS(uint16_t, uint16_t, uint8_t, uint8_t, uint8_t*, uint16_t, uint16_t)':
RX_FSK/src/Display.cpp:379:32: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
          *(uint32_t *)(&tile[0]) = *(uint32_t *)(&(stattiles[stat[i]]));
                                ^
RX_FSK/src/Display.cpp:379:71: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
          *(uint32_t *)(&tile[0]) = *(uint32_t *)(&(stattiles[stat[i]]));
                                                                       ^
RX_FSK/src/Display.cpp:380:32: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
          *(uint32_t *)(&tile[4]) = *(uint32_t *)(&(stattiles[stat[i+1]]));
                                ^
RX_FSK/src/Display.cpp:380:73: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
          *(uint32_t *)(&tile[4]) = *(uint32_t *)(&(stattiles[stat[i+1]]));
                                                                         ^
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/ttgo-lora32/src/src/Display.cpp.o: No such file or directory
*** [.pio/build/ttgo-lora32/src/src/Display.cpp.o] Error 1
============================================================================= [FAILED] Took 7.94 seconds =============================================================================

`

gariac, 2023-05-16:
It doesn't work. Here is the burn:
`
 esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 firmware.bin
esptool.py v3.3.2
Serial port /dev/ttyUSB0
Connecting...........
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:f7:d8:8c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00140fff...
Flash params set to 0x022f
Compressed 1310464 bytes to 756329...
Wrote 1310464 bytes (756329 compressed) at 0x00001000 in 12.9 seconds (effective 810.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

`
To be sure I powered it down and then booted again.

Here is the platformIO compilation output:

`
Processing ttgo-lora32 (platform: https://github.com/platformio/platform-espressif32.git#v3.3.2; board: ttgo-lora32-v1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v1.html
PLATFORM: Espressif 32 (3.3.2+sha.e170b13) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting RX_FSK.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.34.17
|-- AXP202X_Library @ 1.1.3
|-- MicroNMEA @ 2.0.6
|-- ESP Async WebServer @ 1.2.3
|-- GFX Library for Arduino @ 1.1.5+sha.38a28fa
|-- AsyncMqttClient @ 0.8.2+sha.7eacb02
|-- ESPmDNS @ 1.0
|-- SPI @ 1.0
|-- SPIFFS @ 1.0
|-- Ticker @ 1.1
|-- Update @ 1.0
|-- WiFi @ 1.0
|-- Wire @ 1.0.1
Building in release mode

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'scripts/makefontpartition.py'
File "/home/mrg/.platformio/penv/lib64/python3.6/site-packages/platformio/builder/main.py", line 181, in <module>
Compiling .pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/Chasemapper.cpp.o
[... 322 lines cut ...]
                                        ^
Compiling .pio/build/ttgo-lora32/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/main.cpp.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/ttgo-lora32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/ttgo-lora32/libFrameworkArduino.a
Indexing .pio/build/ttgo-lora32/libFrameworkArduino.a
Linking .pio/build/ttgo-lora32/firmware.elf
Retrieving maximum program size .pio/build/ttgo-lora32/firmware.elf
Checking size .pio/build/ttgo-lora32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  37.4% (used 122492 bytes from 327680 bytes)
Flash: [==========]  100.0% (used 1310358 bytes from 1310720 bytes)
Building .pio/build/ttgo-lora32/firmware.bin
esptool.py v3.1
Merged 1 ELF section
============================================================================ [SUCCESS] Took 48.74 seconds ============================================================================
 *  Terminal will be reused by tasks, press any key to close it. 

`

hr-ru, 2023-05-16:
What does "does not work" mean?
Your output seems to show that platformIO does successfully write the firmware to the ESP32

What you don't show is writing the filesystem. Did you do a "pio run --target=uploadfs"?

If you did, what is happening? If you don't see anything happening on the ESP32, look at the serial console.

gariac, 2023-05-16:
I don't have pio cli. I have the ide installed on top of MS "code". You should be able to run the same commands from the IDE except I haven't found where to do so.  Do you suggest I install the CLI?

You can see that I uploaded the firmware using the esptool.py command as the acid test. That is could someone download the file and upload to the board independent of the PlatformIO IDE. And a blank screen.

I decided to go back to the IDE and do the upload from the IDE. That is the right facing arrow at the bottom of the IDE. And the ttgo board boots with the program. However upon close inspection it appears to be the old official build with all my frequencies in the data file. It even has the password for the hot spot from my phone. 

For sanity sake I repeated the esptool,py upload and I got a blank screen. I did the right arrow IDE upload and I have the old code running.  Here is the terminal:
`
 *  Executing task: platformio run --target upload --upload-port /dev/ttyUSB0 

Processing ttgo-lora32 (platform: https://github.com/platformio/platform-espressif32.git#v3.3.2; board: ttgo-lora32-v1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v1.html
PLATFORM: Espressif 32 (3.3.2+sha.e170b13) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting RX_FSK.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.34.17
|-- AXP202X_Library @ 1.1.3
|-- MicroNMEA @ 2.0.6
|-- ESP Async WebServer @ 1.2.3
|-- GFX Library for Arduino @ 1.1.5+sha.38a28fa
|-- AsyncMqttClient @ 0.8.2+sha.7eacb02
|-- ESPmDNS @ 1.0
|-- SPI @ 1.0
|-- SPIFFS @ 1.0
|-- Ticker @ 1.1
|-- Update @ 1.0
|-- WiFi @ 1.0
|-- Wire @ 1.0.1
Building in release mode

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'scripts/makefontpartition.py'
File "/home/mrg/.platformio/penv/lib64/python3.6/site-packages/platformio/builder/main.py", line 181, in <module>
Compiling .pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/Chasemapper.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/DFM.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/DecoderBase.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/DefaultFonts.c.o
Compiling .pio/build/ttgo-lora32/src/src/Display.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/M10M20.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/MP3H.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/RS41.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/RS92.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/SX1278FSK.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/Scanner.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/ShFreqImport.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/Sonde.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/TFT22_ILI9225.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/aprs.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/fonts/fonts.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/geteph.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/json.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/mqtt.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/nav_gps_vel.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/rs92gps.cpp.o
Compiling .pio/build/ttgo-lora32/src/src/rsc.cpp.o
[... 335 lines cut ...]
Writing at 0x000dd833... (61 %)
Writing at 0x000e33fd... (63 %)
Writing at 0x000e9092... (65 %)
Writing at 0x000eed4b... (68 %)
Writing at 0x000f494b... (70 %)
Writing at 0x000fa5a1... (72 %)
Writing at 0x00100527... (74 %)
Writing at 0x00106050... (76 %)
Writing at 0x0010c024... (78 %)
Writing at 0x00112831... (80 %)
Writing at 0x00118590... (82 %)
Writing at 0x0011f137... (85 %)
Writing at 0x0012789b... (87 %)
Writing at 0x0012edda... (89 %)
Writing at 0x00135300... (91 %)
Writing at 0x0013b892... (93 %)
Writing at 0x00143143... (95 %)
Writing at 0x00148f15... (97 %)
Writing at 0x0014eb6b... (100 %)
Wrote 1310464 bytes (756329 compressed) at 0x00010000 in 18.6 seconds (effective 563.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

hr-ru, 2023-05-16:
Also on VSCode you should have some button for "upload file system". Something like this:
[https://community.platformio.org/uploads/default/original/2X/e/ee20eb27230e6d1fb22d183f59b5f36b06458433.png](https://community.platformio.org/uploads/default/original/2X/e/ee20eb27230e6d1fb22d183f59b5f36b06458433.png)

gariac, 2023-05-17:
Ok now THAT work. Thanks hr-ru. It is scanning different frequencies. With luck I can catch the radiosonde that is still flying.

dl9rdz, 2023-08-21:
Alright, seems like this has been solved, so closing this issue...

closed 2023-08-21 by dl9rdz

gariac, 2023-08-22:
Ah if I was supposed to close it myself, my apologies.

I learned quite a bit though in the thread. I have a feature suggestion, but that will be for a new issue.