dl9rdz/rdz_ttgo_sonde #367: remove the "BACK" button

RV88470, 2023-07-08:
Hello, how to remove the "BACK" button from the Livemap screen?
http://rdzsonde.local/livemap.html
THANKS

dl9rdz, 2023-07-09:
Hi,
You would have to edit the livemap.js on the local file system on the TTGO and remove this line:
map.addControl(new L.Control.Button([ { position: 'topleft', text: '🔙', href: 'index.html' } ]));

Best,
Hans

RV88470, 2023-07-09:
Thank you very much for your response and help.
I can edit the file, but when I want to save the changes, I get this error message :

ERROR
Something went wrong (probably ESP32 out of memory). Uploaded file is empty.

J'utilise ce récepteur : 
[(https://fr.aliexpress.com/item/32872078587.html?spm=a2g0o.order_list.order_list_main.16.48425e5bQRBfFx&gatewayAdapt=glo2fra)](url)
It seems that the memory is insufficient

Beautiful day
Hervé

hr-ru, 2023-07-09:
I assume you are trying to modify the file via edit.html?file=... 
The file is too large for this method.

You need a different approach to edit the file. Many possibilities
- Use something like `curl http://rdzsonde.local/file/ --upload-file livemap.js` (not sure about the exact parameters, no device at hand to test)
- Use the ttgoconfig script, `scripts/ttgoconfig put file livemap.js` (livemap.js needs to be in the current directory)
- Upload the whole file system using either the Arduino tool or the platform.io tool (see wiki how to use)

RV88470, 2023-07-09:
Thank-you for your prompt response. I will try that soon.
Good Sunday

closed 2024-01-08 by dl9rdz