dl9rdz/rdz_ttgo_sonde #368: RS92 is not decoded

Mauri740, 2023-07-14:
Hi, there are some Rs92s in the air but no correct decode data from the software. I updated with V0.9.3 but the problem is the same. I think that the cause is the offline server (gssc.esa.int/gnss/data/daily/) that not allowing download of eph data.  Can solve this issue? It would be nice for me to recover an old Rs92. Thank you

dl9rdz, 2023-07-14:
Well, yes, ftp sources are gradually disappearing.

Currently there is no space for TLS in the firmware, but the current plan (next devel) is to enable http download, and have a http=>https running (soon in the next devel firmware)

But for now, the esa ftp server seems to be still available.
```
~$ ftp gssc.esa.int
Connected to gssc.esa.int.
220 Wing FTP Server ready...
Name (gssc.esa.int:hans): anonymous
331 Password required for anonymous
Password:
230 User anonymous logged in.
Remote system type is UNIX.
Using binary mode to transfer files.
ftp> cd /gnss/data/daily/2023/195
250 CWD command successful. "/gnss/data/daily/2023/195" is current directory.
ftp> ls br*
200 Port command successful
150 Opening data channel for directory list.
-rw-rw-r-- 1 5510 gssc_adm          49222  Jul 14 18:03 brdc1950.23n.gz
-rw-rw-r-- 1 5510 gssc_adm          58631  Jul 14 18:03 brdc1950.23g.gz
226 Transfer successful.

```

And my test TTGO just now successfully downloaded eph data with the standard config. Maybe the server was temporarily unavailable, or you are behind a firewall that is blocking ftp connections?

Mauri740, 2023-07-14:
Thanks for your quick reply. I'll check if my firewall is blocking ftp system. Now I have inserted a free line with type RS92 in the QRG list and a second one with sure frequency for the Rs92 sonde. I checked the firmware information and it show:
RS92 RINEX eph state: Good
Could that be enough to receive a Rs92?
I will wait patiently for the new version to fix this issue.
Thank you for your work and your help.
Greetings

dl9rdz, 2023-07-14:
Yes, if it says Good, then it successfully fetched eph data from the ftp server.

Note that fetching eph data is done only once, right after startup.
This is good for mobile use, but now more and more people are using the TTGO in a fixed location running continuously. You have to reboot manually to fetch new eph data (accuracy degrades if it is older than approx. 6h)

Mauri740, 2023-07-14:
Thank you so much for support. The situation begins to be clearer. I hope to receive a Rs92 as soon as possible even if they are very rare. Below is a screen shot of the data as it was received a few days ago from one of the three sondes flown nearby. As you can see, the Qth is wrong like the altitude and ID.  These are the three IDs: 
L2233714 
K3523162
K3424059
none of the three were received regularly, perhaps because the eph data was "too old"
[image]

Mauri740, 2023-08-17:
Now it seems that the server, or something connected to it, is down. No Rinex eph data is downloaded. I have three stations with rdz and they all show "fetch error" or "disabled".  Could it be the end of the ftp connections?

14ri004, 2023-08-26:
Hello hansi, we have the same problem, impossible to decode RS92 ---> RS92 RINEX eph status: Disabled

FTP server for ephemeris data (RS92 decoder) = gssc.esa.int/gnss/data/daily/

dl9rdz, 2023-08-26:
Yes, well, the server is still, but a couple of days back it stopped providing the brdc...gz file. I have not found any good replacement ftp server. So I guess I should switch over to http oder https soon...

```
ftp> ls daily/2023/223/brdc*
200 Port command successful
150 Opening data channel for directory list.
-rw-rw-r-- 1 5510 gssc_adm          25479  Aug 11 08:37 brdc2230.23n.gz
-rw-rw-r-- 1 5510 gssc_adm          27647  Aug 11 08:38 brdc2230.23g.gz
226 Transfer successful.
ftp> ls daily/2023/224/brdc*
200 Port command successful
150 Opening data channel for directory list.
226 Transfer successful.
ftp> 
```

14ri004, 2023-08-26:
neither of the 2 FTP addresses works, you tested with which FTP address ????

gssc.esa.int/gnss/data/daily
igs.bkg.bund.de/IGS/BRDC/

dl9rdz, 2023-08-26:
Well, I can connect to the esa server, but as I said, the server does not provide the files. So it is not working.

14ri004, 2023-08-26:
ok merci

dl9rdz, 2023-08-28:
Actually, brdc files are on the server, just in a different location than before.
devel2023-08-28 fixes that.
The config format changed, now it is the full path/format string to the file (should update automatically)
gssc.esa.int/gnss/data/daily/%1$04d/brdc/brdc%2$03d0.%3$02dn.gz

14ri004, 2023-08-28:
Hello Hansi, even after the update RS92 RINEX eph state: Disabled
the FTP address is up to date as you indicate, but RINEX still out of order.
I added 2 RS92 frequencies and rebooted the TTGO

gssc.esa.int/gnss/data/daily/%1$04d/brdc/brdc%2$03d0.%3$02dn.gz

14ri004, 2023-08-28:
[image]
[image]
[image]

dl9rdz, 2023-08-28:
Did you reboot? 
If yes

It works here on my test board...

14ri004, 2023-08-28:
YES I rebooted 3 times, turned off / on again, always the same and this on 3 TTGO and 2 T-BEAM

dl9rdz, 2023-08-28:
I don't see any way how this can happen. But maybe you could send me the output from the serial console?

14ri004, 2023-08-28:
will you have to tell me how to do it?

dl9rdz, 2023-08-28:
Board needs to be connected using USB.
Then, depends on what do you usually use to program the board?

- Arduino IDE: Press the Serial Monitor icon on right top.
- PlatformIO command line: "pio device monitor" (run in the same folder as were the platfromio.ini file is)
- PlatformIO/VSCode: Don't remember, don't have it installed right now, but there was some serial monitor button somewhere as well.
Or, as an alternative, any serial console program (select the right serial port device, and baud 115200) should work as well.

Mauri740, 2023-08-28:
Hello Hansi, I updated my Ttgo with devel2023-08-28 and now I have Rs92 Rinex eph state: good. thank's for your job!
greetings
[image]

dl9rdz, 2023-08-28:
Basically the functionality is simple:
- After booting, if there is no channel (frequency) with RS92, the state is set to disabled (and remains disabled as long as the TTGO is running)
- Otherwise, the software tries to fetch eph, and then the state is something different in all cases. You should never see "disabled" in that case. The only possibilities are either "good" or some error state, but never "Disabled".

Seems like you have a slightly modified (french) version of the software, maybe something is broken in that version? Or did you try my original version as well? The web content in data does not matter, but the word "Désactivé" is printed in the software.

Otherwise it is very strange and I need the serial port output to diagnose this further.

14ri004, 2023-08-29:
yes I put 3 frequencies with RS92, I updated with the last version of the software (version devel20230828b), no modification of the software either, reboot, turning off and turning on the ttgo does not change anything,
ftp=gssc.esa.int/gnss/data/daily/%1$04d/brdc/brdc%2$03d0.%3$02dn.gz

Mauri740, 2023-08-29:
could this help you?
[image]

14ri004, 2023-08-29:
[image]
[image]

14ri004, 2023-08-29:
I is not the complete line in the FTP ??? I'm missing information

gssc.esa.int/gnss/data/daily/%1$04d/brd

gssc.esa.int/gnss/data/daily/%1$04d/brdc/brdc%2$03d0.%3$02dn.gz

14ri004, 2023-08-29:
I'm not the only one in this situation, I have 2 friends who have exactly the same problem = RS92 RINEX eph state: Disabled

dl9rdz, 2023-08-29:
Ok, there is an issue with the config editor. But this should only case "Fetch error", never "Disabled". 

What you can do is just remove the geteph url line in the config and save, and then reboot, then it will use the correct default string automatically (this problem is fixed in new devel20230829).

But please try to get the serial port output. If you cannot use Arduino or PlatformIO, you can also simple Windows commands. Use device manager to find the com port number of your ESP32 (connected via USB), and then open the command line and do
`TYPE COM3: >> serial.txt`
Instead of "3" use the correct com port number :)
You can also send the file to me by email instead of posting it here (dl9rdz@darc.de)

14ri004, 2023-08-29:
I reprogrammed 2 new TTGOs with: flash_download_tool_3.9.4.exe
always the same result = disabled
I don't use Arduino or platfomIO, I tested your command (TYPE COM14: >> serial.txt) but it tells me access denied,

Mauri740, 2023-08-29:
Try to upload the official version "master v0.9.3.B17" here: http://rdzsonde.mooo.com/download.html
Then, from the "ABOUT" Ttgo page, update to "devel 20230828" via "check for update"
If the ttgo board is the same to mine, as it's the firmware version, I don't understand how there can be a different result.

14ri004, 2023-08-29:
j'ai fait la mise à jour en master et j'ai remis la devel, exactement pareil, j'ai tester sur 7 TTGO dont 3 T-BEAM résultat identique, j'ai mis 3 fréquences en RS92, reboot, effacer la ligne FTP, éteindre, allumer, cela ne change pas, c'est toujours en désactivé, a ne rien y comprendre

dl9rdz, 2023-08-29:
Don't do the "cd .."   You probably don't have permissions to write to the root directory, but you should be allowed to write to your home directory

14ri004, 2023-08-29:
ne fonctionne pas hansi, désolé

dl9rdz, 2023-08-29:
What if you do "TYPE COM14:" (without the ">> serial.txt)

14ri004, 2023-08-29:
[image]
[image]

I tried everything, does not work, I put another TTGO = com32

dl9rdz, 2023-08-29:
So this should be like this, and it works here
[image]

- TTGO must be connected
- Check COM port number in device manager to make sure to get the right one (maybe it changes?)
- TYPE COMxx: in shell

I

dl9rdz, 2023-08-29:
Or, my last attempt, you can try this:
- Install Arduino IDE 1.8.9 (from https://www.arduino.cc/en/software)
- Connect TTGO board
- Run Arduino IDE
- Go to "Tools" -> "Port" and select TTGO COM port
- Press icon on right top corner (serial monitor)
- A window should pop up. Probably printing garbage. On the bottom, where there is "9600", change baud to "115200"
- Press Reset on the TTGO
- Wait until TTGO has finished startup
- Then you should have a lot of text in the window. You can copy all the text with Ctrl+A Ctrl+C

Mauri740, 2023-08-29:
I just updated my third TTgo too, everything worked. Rinex: Good. 
[image]

14ri004, 2023-08-29:
YES it's good hansi, bravo, I gave you a lot of work, now it works, thank you.

[image]

[image]

dl9rdz, 2023-08-30:
Alright, and just right now seems like ESA has changed the server back to the previous structure. 

So now you need to adjust the configuration again to this one:
gssc.esa.int/gnss/data/daily/%1$04d/%2$03d/brdc%2$03d0.%3$02dn.gz

14ri004, 2023-08-30:
hello hansi, yes it's done and everything works now, it was only the wifi mode (3) which was blocked, thank you for your help, have a good day.

14ri004, 2023-09-20:
Hello Hansi, I'm coming back to you, to tell you that RS92 is still undecodable with TTGO, only the auto-RX manages to receive data, for my part I receive frames but no valid positions, see screenshot , CDT

[image]

hr-ru, 2023-09-20:
Do you have ephemeris data on the TTGO (check in the "About" tab of the web interface)?

14ri004, 2023-09-20:
YES hansi, Wifi in 3, rinex good, frequency OK, I receive frames, but no decoding of RS92