dl9rdz/rdz_ttgo_sonde #369: Incorrect APRS Format packet being generated

ab0oo, 2023-07-29:
https://github.com/dl9rdz/rdz_ttgo_sonde/blob/27d342ae0e8a184b91a063b25c2367e34c5316c9/RX_FSK/src/aprs.cpp#L279

Saw this packet come across APRS-IS tonight:
`ON2KGC-5>APRRDZ,TCPIP*,qAC,T2BELGIUM:!5035.56N00553.63EON2KGC-13 rdzTTGOsonde-devel20230427
`
it looks to be invalid and unparsable.  !5034.56N  <Symbol Table Identifier Missing>00553.63E <Symbol Code missing> ON2KGC-13.....
Should be a pretty quick fix. I see another radiosonde tracker:
`PANHA2-15>APNL51,TCPIP*,qAI,PANHA2-15,SQ6KXY-1,EIGHTH,NINTH:!4740.10N/01738.11E`&#96;`Radiosonde Tracker - Based on kxyTrack 1.5.2-dxl_smod136h
`
using / for the symbol table and  &#96; for the Symbol.

hr-ru, 2023-07-29:
Ok, thanks.
This seems to be more a configuration problem than a software problem.
I assume that the symbol table and symbol is empty in the sender's configuration.
But maybe I can add some check and make the software use a default symbol if the configured symbol is invalid...

ab0oo, 2023-07-30:
Ah, I didn't look to see  that it was user-configured.  Just an education issue more than a code issue.  abouit 4% of the packets on APRS-IS are malformed, so this is a drop in a very deep bucket.

dl9rdz, 2024-08-12:
Finally added this check to the code, will be available in the next dev release, so I expect that we will not see this issue with that version any more. Closing this isse...

closed 2024-08-12 by dl9rdz