dl9rdz/rdz_ttgo_sonde #371: Disable AP functionality and make reconnect to wifi more robust

j0uni, 2023-08-01:
It seems that if rdz boots up before wifi station is ready, rdz will go to AP mode and is out of network until physically rebooted.

I would add two features:
- disable AP feature and make rdz to search for networks until it finds one in boot (to make sure it comes back to network after power outage)
- make wifi connection lost more robust (rdz would try to connect back to any predefined wifi networks if connection is lost)

These ideas come from experience where after power outage rdz boots faster than the wifi router.

dl9rdz, 2023-08-01:
Did you check Wifi mode 1 (instead of the default mode 3: automatic choice between station and AP mode)?

(1: Wifi station mode. Connects to AP in background and dynamically obtains an IP address (connection ok: IP address shown in display). Connection is made to a network configured in the WiFi tab/network.txt with best RSSI value. Automatically re-connnects in case of connection failure.)

Might be the thing you want?

j0uni, 2023-08-01:
Ah, I have to check that it is in mode 1 when I get it into network.