dl9rdz/rdz_ttgo_sonde #372: WiFi password length problems.

JDat, 2023-08-03:
I have 30 character long WiFi password on my router.
Looks like RDZsonde code support only up to 8 characters.
Can you increase WiFi password length to 30 characters?

dl9rdz, 2023-08-03:
No, there is no limitation in password size in my code.
(ESP32 code might limit password length to 32 characters I think)

dl9rdz, 2023-08-21:
Just as a quick follow-up, maybe another possibilityis that you have some special characters in the password that are not handled correctly by the web interface. You could try editing the password file directly (Arduino or PIO file system upload, or http://rdzsonde.local/edit.html?file=networks.txt)

JDat, 2023-08-23:
No, I am using only Latin alphabet and numbers. Just turned on, got latest update. Looks like everything working. We can close issue.

dl9rdz, 2024-01-07:
Closing this issue. The other problem with not properly quoting " chars in the password (in the web interface) has also been fixed now.

closed 2024-01-07 by dl9rdz