dl9rdz/rdz_ttgo_sonde #378: Add support for AXP2101 (battery voltage + 2nd button)

SzczepanLeon, 2023-08-23:
New T-Beam V1.2 has replaced PMU

There is no AXP192 but AXP2101

Please add support for that chip, probably migration to [XPowersLib](https://github.com/lewisxhe/XPowersLib) is needed.

tomtom543, 2023-09-21:
Hi, just wondering if you have seen this https://www.aliexpress.com/item/4001178678568.html in which they say that adding a 22R resistor between GPS Supply rail and LORA Supply rail fixes the low output from the new power chip that causes the LORA chip not to work? It could get it working until the software can be updated.

SzczepanLeon, 2023-09-21:
Interesting. There is nothing about SW loaded. What version erc. Maybe this is dirty HW hack for SW without AXP2101 support?

dl9rdz, 2023-09-21:
Well, devel20230828b should contain all necessary code for (a) initializing the AXP2101 and (b) handling the AXP2101 button. Let me know if not. There is also a pull request with updates that add support for reading battery values from the AXP2101 that I still have to merge.

The site you are linking to shows that this is from Aug 25, so that was before the devel of Aug 28 was released :) So at that time such mod might have been necessary.

SzczepanLeon, 2023-09-21:
> So at that time such mod might have been necessary.

Yes, I think so.

Hack HW or write copule lines of code ;)

dl9rdz, 2023-10-19:
No update since devel20230828b, and with that:
- The power configuration is working, i.e. RX and GPS should be working fine
- Enhancements on the battery stats on the screen are in a pending pull requests. Still have to merge that (Sorry, I was too busy over the last weeks)
- The AXP2101 is different and does not provide all features, in particular current measurements are not supported. So these will never be working in the same way as with older T-Beam.

SzczepanLeon, 2023-10-19:
I can confirm rhat Rx & GPS are working on that dev.

With that power chip we don't have info about current(s) but battery precentage is available.

rafalantas, 2023-10-19:
did you configure anything special for that GPS ? i tried to get fix for 30 mins outdoor and nothgin :(

SzczepanLeon, 2023-10-19:
> did you configure anything special for that GPS ?

Nothing special. Pins are default. Power as on dev.