dl9rdz/rdz_ttgo_sonde #380: frequency-error in the webinterface?

DG0MG, 2023-08-26:
[https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Receiver-configuration](url)

I would like to have the value of the frequency-error (shown on the Display) viewable somewhere in the webinterface too. 
Perhaps on the DATA-Tab, separately for every received Sonde.

The reason: I have the TTGO in the attic and there is a certain temperature difference between summer and winter, which causes a frequency offset of a few kHz. To check it, I have to climb up to the attic every time to read the display.

[image]

14ri004, 2023-08-28:
hello, you put your IP address of your TTGO +: 14570 (

http://192.168.x.xxx:14570/

= AFC xxx (you will have the information you are looking for)

[image]

dl9rdz, 2023-08-28:
Well, you can do that (but that port is for the app that will keep the connection open, not the best for the browser), yet to better version of this is to use http://rdzsonde.local/live.json  (or with the IP, but instead of :14570, just add live.json

DG0MG, 2023-08-28:
okay, thank you both for the hints - this solution is enough for me to occasionally check the frequency drift.

[image]

73-de-LZ, 2023-09-06:
You can try this:
[Blue Dot Mod](http://lz4tu.freevar.com/RDZmods/RDZmodsIndex.html)
[image]
73!