dl9rdz/rdz_ttgo_sonde #384: Meteomodem M20 serial decoding sometimes incorrect?

Eshco93, 2023-10-02:
Hello,

I recently dug a little deeper into the radiosondy.info database and I noticed something that might be an issue.
Please take a look at these two radiosondes:

- [MEA061741 (209-2-61741)](https://radiosondy.info/sonde_archive.php?sondenumber=MEA061741)
- [MEA012637 (209-2-12637)](https://radiosondy.info/sonde_archive.php?sondenumber=MEA012637)

When comparing the data, I'm under the impression that those were not two different radiosondes. It seems to me that this was just a single radiosonde, but different stations with different softwares decoded different serials and therefore two different entries were produced in the database.

MEA061741 (209-2-61741) was exclusively decoded by stations using rdzTTGOsonde, while MEA012637 (209-2-12637) was decoded by a station using radiosonde_auto_rx.

Here is another example where this happened:

- [ME2862173 (305-3-62173)](https://radiosondy.info/sonde_archive.php?sondenumber=ME2862173)
- [ME2813068 (305-3-13068)](https://radiosondy.info/sonde_archive.php?sondenumber=ME2813068)

I'm not quite sure which software decodes an incorrect serial and which serial is actually correct. But I suspect that rdzTTGOsonde is decoding the incorrect one.

Greetings,
Simon

hr-ru, 2023-10-02:
Thanks for reporting this. Actually I see two separate issues:
- There is a bug in my software (signed vs unsigned) causing a wrong value to be printed. It *should* print the same Id as computed by dxlaprs, but due to this bug it does not. With that bug corrected, it would print 209-2-10829 for the first sonde.
- There is a difference between how auto_rx and how dxlaprs calculate M20 IDs. For IDs nnn-n-1nnnn there is a differences of 1808 (10000 - 8192). Do you have some evidence that the auto_rx version is the correct one here? If yes I can surely switch to that version, but we should also inform Chris.

Eshco93, 2023-10-02:
Thank you for the quick reply.
I actually don't have any evidence that the auto_rx serial is the correct one here. Like I said, that's just a guess.
But I will try to get this evidence. My idea would be to search the database for M20 radiosondes where this issue could be observed. Maybe some of those radiosondes where found and maybe I can reach out to the finders and ask them what the correct serial is.
I'll let you know as soon as I have any information about it.

Eshco93, 2023-10-03:
I have some new information.

I talked to Yannick Saindrenan (F5IKO) from France, who found the second radiosonde that I mentioned in my initial post. That radiosonde actually isn't an M20. It's a PS20 pilotsonde, which is actually a stripped down version of the M20 with no sensor arm, no parachute and no serial number sticker.
He claims that ME2813068 (305-3-13068) is the correct serial number. But I'm not quite sure where this information comes from. When you can't trust your decoder and there is no sticker on the radiosonde itself - well...how do you know it then?
Maybe I'm missing something here. I asked him how he knows which serial number is correct. And I'm waiting for his reply.

Vigor G. - another guy from France that I recently came into contact with - is also investigating this issue. He contacted Meteomodem and they told him that M20 production currently doesn't exceed 8192 units per month. And since that last part of the serial number is a unit number which is assigned in ascending order beginning at zero for every new month, it's safe to say that this number should be less than 8192. Otherwise it's not valid or it's not an M20.

dl9rdz, 2023-10-03:
Alright, as long as the last 5 digit number is below 8192, all decoders will produce the same serial number :)

The interesting part is when it reaches that limit. After 8191, dxl will simply continue with 8192, but auto_rx will continue with 10000. 

Another unclear part is the "middle" number. For -2- and -3- seems like all software produces the same number, but auto_rx will never output -1- (the code is simply doing 2 + (some value 0..3)). The information I got is that the number indicates the production site (with values 1/2/3), but I have seen -1- only in old instances from 2019.

Eshco93, 2023-10-08:
Vigor said that factory number 1 was from a previous factory that doesn't produce Meteomodem radiosondes anymore.
So it's quite likely that we'll never see 1 again.