dl9rdz/rdz_ttgo_sonde #386: rdz_ttgo_sonde master v0.9.3 GPS Problem

eb2fiz, 2023-10-25:
Hi.
I have been using this software for years and I have noticed that after updating it to the master version v0.9.3 the GPS in the application does not update automatically. To get a real value, I have to disconnect the GPS on the phone and reconnect it. The tests have been done by retrieving RS-41 probes.

Thanks and great work.

dl9rdz, 2023-10-25:
Hi,

can you please explain in more detail? 
"the GPS": are you talking about the GPS position of a RS41, of the internal GPS of the TTGO board, or your phone's GPS position?
"the application": what does not update? The position (of a probe? of yourself?) shown on a display connected to the TTGO (LCD or TFT)?  The position shown on the web interface of TTGO (accessed by your computer or phone)?  The position shown in the rdzwx-go app on an Android phone?

Also, what do you mean with "disconnecting the GPS on the phone"?

Thanks.

eb2fiz, 2023-10-25:
Hi.

It does not update its own position relative to the sonde on the map in the drzwx-go application on the phone. Then I disable the GPS on the phone and when I re-enable it, it gives me my rela position in the drzwx-go application on the phone.
The reception of the sonde on the TTGO and in the drzwx-go application is perfect.

Thanks.

dl9rdz, 2023-10-25:
Alright, thanks.

I think this is a problem I solved back in August in the devel version. 
https://github.com/dl9rdz/rdz_ttgo_sonde/commit/082b6ccdf54e494e03a8650d2aee4277277da57d

Can you try the latest devel version and check if that solves the problem? The next master version will then incorporate these changes as well..

eb2fiz, 2023-10-25:
Perfect.

I'll put the August version and after testing it I'll let you know.

Thanks.

eb2fiz, 2023-10-25:
Solved !!!!

I've updated to the August version you mentioned, I've tested it and it works fine.

Thanks and best regards.

closed 2023-12-03 by dl9rdz