dl9rdz/rdz_ttgo_sonde #393: NMEA frames

14ri004, 2023-11-07:
hello hansi, I would like to know how to retrieve NMEA frames on a TTGO RDZ, and be able to receive them on my phone, thank you

hr-ru, 2023-11-07:
NMEA from the TTGO's internal GPS?

There is no such functionality (as direct NMEA frames).
You can fetch http://rdzsonde.local/live.json and there you will have the position reported from the TTGO's GPS.

Or do you mean the decoded sonde's position in NMEA format?

14ri004, 2023-11-07:
yes, what I want is to be able to recover the position of the probe, from my TTGO in decoding, what possibilities are there??

hr-ru, 2023-11-07:
Well there are many ways... You want to implement your own software? Or use the phones web browser? or some existing app on the phone?

hr-ru, 2023-11-07:
In more detail:
- There is the internal web server providing probe location on http://rdzsonde.local/live.json (that is used internally on the TTGO web interface map) and http://rdzsonde.local/status.html (used by the "Data" tab of the web interface). You can directly access these two with your web browser as well (or query in your own app)
- There is also a kml file provided by the web server at http://rdzsonde.local/dynamic.kml. Some map apps (e.g., Locus map) can use that to import the probe's positions (some even with automatic dynamic updates)
- There is a TCP socket interface on port 14570. This is used by my Android app (rdzwx-go). The port sends you all positions updates in JSON format (the main difference to live.json is push vs. pull model)
- There is the AX25/TNC via TCP interface. The APRSdroid app (and possibly other APRS apps) use it. 

(There are more options such AXUDP and MQTT but these are probably less suitable for what you want to do)

14ri004, 2023-11-07:
thank you both for the information provided, I will see what can be done, best regards

closed 2023-11-07 by 14ri004