dl9rdz/rdz_ttgo_sonde #394: Order on Data page constantly changes

m-ugo, 2023-11-09:
The order of received data change randomly on the Data page every 5 seconds when the page is refreshed. If it is sorted by latest received data or same order as on the QRG page doesn't matter as long as it doesn't change every refresh.

Thanks for a nice piece of software! :-)

hr-ru, 2023-11-09:
Its not completely random, its just starting with the frequency entry currently displayed on the TTGO display on top, and then all subsequent entries following the order in the QRG page, wrapping around to top when reaching the bottom. 

When the TTGO is receiving something, this is fine.  
But I agree that this is not a good design when the TTGO is in scan mode.. Its on my low-priority TODO list to change that somehow. Sorted by lasted received data probably is a good choice...

Medvidas, 2025-02-10:
I have touched on the same topic here, with a possible outline of the look:
https://github.com/dl9rdz/rdz_ttgo_sonde/discussions/400

The Data page in its current state is hardly usable, while constantly rotating  it is
almost impossible to copy coordinates etc. It will be nice to see some slight revamp of it.

What more, it will help to include and visualise the Launch location in the table.