dl9rdz/rdz_ttgo_sonde #395: Non Broadcasting Wifi

skysho, 2023-11-10:
I recently changed my wifi to no longer broadcast and two of our RDZ's failed to connect.  I went in and made sure the information was accurate and even deleted and re-added the SSID and PW.  Still no luck, had to turn back on the broadcasting.  Two of my RPi running auto_rx and the other Radiosondy both dropped off but one of them had the ability to connect to a non broadcasting wifi.  So it worked fine, the Sondehub RPi didn't give me a choice so I cabled it to my switch.  Thanks for any suggestions.

hr-ru, 2023-11-10:
What WiFi mode are you using?

hr-ru, 2023-11-10:
In any case, I just checked, it will not work with hidden APs, as the software always uses WiFi scan (and this will not find non-broadcasting APs).  It would be possible to change to code such that it works with hidden APs, but this is currently not implemented.

skysho, 2023-11-10:
Thanks for the info, I'll quit trying to find a work around.

skysho, 2023-11-10:
Sorry, its just an Xfinity router so I’m not quite sure about the mode.

From: Hans P. Reiser
Sent: Friday, November 10, 2023 2:42 PM
To: dl9rdz/rdz_ttgo_sonde
Cc: skysho; Author
Subject: Re: [dl9rdz/rdz_ttgo_sonde] Non Broadcasting Wifi (Issue #395)

What WiFi mode are you using?

dl9rdz, 2024-01-07:
Today's devel version adds a new WiFi mode 4. 

This mode directly connects to the second network in the config (i.e., # 1 in the web interface, below the AP name). It does not scan, so it works also with hidden SSIDs.

Note that in this mode you cannot configure multiple other networks.

Another change: In mode 1, 3 and 4, during the initial network scan, you can press any button of your board to change to AP mode temporarily (until reboot or rescan).

closed 2024-01-07 by dl9rdz