dl9rdz/rdz_ttgo_sonde #402: TTGO TBEAM axp2101 V1.2 BAttery Chargeing

sivaelid, 2023-11-21:
**Describe the bug**
 The Issue is that while in the latest  dev firmware does enable the powing of the  Lora and GPS modules  The charge circuit and battery management  do now work. It is seen form otehr bug reports that there are still issues with the APX2101 PMU..

Looking at otehr uses of the T-Beam v1.2 such as TTN they have incorporated the Library and   the configuration is seen in posts https://www.thethingsnetwork.org/forum/t/new-t-beam-version-1-2/63918/10 with specifically  configuration of the charge setting to be 
// Set constant current charge current limit
        PMU->setChargerConstantCurr(XPOWERS_AXP2101_CHG_CUR_500MA);

        // Set charge cut-off voltage
        PMU->setChargeTargetVoltage(XPOWERS_AXP2101_CHG_VOL_4V2);
I hope above helps

**To Reproduce**

Use  TTGO T-Beam  and  plug in power supply to Micro USB and power board . A blue flashing light ( Charge Light is  flashing) but it never put charge in the battery,  and the battery will go flat if used. The battery can not be charged back up.

**Expected behavior**
 That the with the power supply connect to a 5v Micro plug and pluged into the board  that the battery voltage  will be charged  with in the limits of the settings above.

**Version and specific configuration:**
 - (version devel20230829)

**Additional context**
 There are otehr issues noted elsewhere where  the Voltage measurment and E Gauge are missing  apart form that the board work in the basic settings.

sivaelid, 2023-11-21:
Additional information . The control registeres need to be defined in  PMU.cpp, PMU.h and set to be used. these register are not used in either the current dev or the pull request (SzczepanLeon) for the AXP2101 that set up reading data. 
The registers:  
Precharge current limit 0x61 is default to 125mA this is ok ,
Charging termination current limit 0x62 is set to default 0 mA ( No charge) needs to be typically set to 500mA Code 01011, Charging termination current limit 0x63 is set to default 125mA

Great work  Simon.

sivaelid, 2023-12-03:
> This is(was) more tricky than expected, but seems to be fine in my test setup now: All those registers you mentioned would work with their default values as well, these are not the reason for the T-Beam not charging the battery. The important part to make it actually work is to disable the TC ADC channel in register 0x30!
> You can connect an external temperature sensor for the battery to TC.  With the measurement enabled (which is the case by default), the AXP2101 assumes it can measure the battery´s temperature. Without the right sensor (which the T-Beam 1.2 does not have), this causes the AXP2101 to enter an error state (abnormal battery temperature)

Thanks for that I will look into the temperature thermistor issue I did not spot that. Look forward to updated code. Have a good week. Simon G8HAMSent from my iPhone

dl9rdz, 2023-12-03:
Should be fixed in the latest devel release (20231203).

closed 2023-12-03 by dl9rdz