dl9rdz/rdz_ttgo_sonde #407: Frozen display screen ILI9225 - T-beam V1.2 - dev soft.

PiroRiroPL, 2023-12-07:
Hello Guys :) 

I'm using a t-beam v1.2 board and i have see/found a bug.
When i'm use ili9225 disp with 180 degree rotated disp (option 3) then when i clik some of buttons on the bottom of board 
to switch type of radiosonde or screen, the disply its frozen and i cant nothing do. Only restart works, but if i clike user buttons again 
the screen froz again and again. 

The problem i have with devel20231203-B17-full.bin and older dev soft. With master release evrything works fine.

Best redgards PiroRiro

Sorry for my bad english :/

PiroRiroPL, 2023-12-07:
Log from ttgo after reboot and clicking middle button:
```
Board fingerprint is 23
Autoconfig: looks like T-Beam 1.0 or M5Stack Core2 board
no I2C display found, assuming large TFT display

... with large TFT display

Initializing SPIFFS
Reading initial configuration
configuration option 'wifi'=3 
configuration option 'mdnsname'=rdzsonde 
configuration option 'ephftp'=gssc.esa.int/gnss/data/daily/%1$04d/brdc/brdc%2$03d0.%3$02dn.gz 
configuration option 'debug'=0 
configuration option 'maxsonde'=20 
configuration option 'rxlat'= 
configuration option 'rxlon'= 
configuration option 'rxalt'= 
configuration option 'screenfile'=2 
configuration option 'display'=0,1,2,3,4 
appending value 0  next is 1,2,3,4
appending value 1  next is 2,3,4
appending value 2  next is 3,4
appending value 3  next is 4
appending value 4  next is 
configuration option 'dispsaver'=0 
configuration option 'dispcontrast'=-1 
configuration option 'norx_timeout'=20 
configuration option 'tft_orient'=1 
configuration option 'spectrum'=-1 
configuration option 'startfreq'=400 
configuration option 'channelbw'=10 
configuration option 'marker'=1 
configuration option 'noisefloor'=-125 
configuration option 'freqofs'=0 
configuration option 'rs41.agcbw'=12500 
configuration option 'rs41.rxbw'=6300 
configuration option 'rs92.rxbw'=12500 
configuration option 'rs92.alt2d'=480 
configuration option 'dfm.agcbw'=20800 
configuration option 'dfm.rxbw'=10400 
configuration option 'm10m20.agcbw'=20800 
configuration option 'm10m20.rxbw'=12500 
configuration option 'mp3h.agcbw'=12500 
configuration option 'mp3h.rxbw'=12500 
configuration option 'call'=N0CALL 
configuration option 'passcode'=12345 
configuration option 'kisstnc.active'=1 
configuration option 'axudp.active'=1 
configuration option 'axudp.host'=192.168.42.20 
configuration option 'axudp.port'=9002 
configuration option 'axudp.highrate'=1 
configuration option 'tcp.active'=0 
configuration option 'tcp.timeout'=0 
configuration option 'tcp.host'=radiosondy.info 
configuration option 'tcp.port'=14590 
configuration option 'tcp.highrate'=20 
configuration option 'tcp.objcall'= 
configuration option 'tcp.beaconsym'=/`/( 
configuration option 'tcp.chase'=0 
configuration option 'tcp.comment'= 
configuration option 'mqtt.active'=0 
configuration option 'mqtt.id'=rdz_sonde_server 
configuration option 'mqtt.host'= 
configuration option 'mqtt.port'=1883 
configuration option 'mqtt.username'= 
configuration option 'mqtt.password'= 
configuration option 'mqtt.prefix'=rdz_sonde_server/ 
configuration option 'cm.active'=0 
configuration option 'cm.host'= 
[... 211 lines cut ...]
MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 129412, unused stack: 4872
PMU_IRQ is set

Keypress: 1 (short)Key 2 was pressed [5]
Sonde:receive(): result 1 (RX_TIMEOUT), event 05 => action 40
waitRXcomplete returning 4001 (RX_TIMEOUT)
Setting display mode 2
setLayout: 2 (max is 15)
MAIN: loopDecoder: action 40 (Display=64) => 255  [current: main=11, rxtask=11]
last_in - now: 272696
MAIN: updateDisplay started
Drawing circle with size 111 at 0,22
GPS0: NCS N=0, A=0, B=-1
25: 55,29
Drawing circle with size 111 at 104,22
GPS0: CCS N=360, A=0, B=359
30: 159,29
MAIN: updateDisplay done (after 34 ms)

``` 
Sorry i have make bad description :/
The display with master release works fine. 
Yes, the pwr button doesn't work, but if i click him the display goes frozen. 
If i click user defined button disp goes frozen too.
To power lora module i have make a wiring between gps vcc and lora vcc with 22R resistor.

hr-ru, 2023-12-12:
Sorry, I cannot reproduce this here, I tested it with all my displays, including two different ILI9225.

Can you describe more details of the problem?
- "disp goes frozen": So it is only the display? The web interface is still working, and you still continue to see output on the serial port? 
- In the log you first pressed the right button, then the left button which activated the "PeilungTFT" screen. Did you see this screen (the two large circles on the screen)?  Does it get stuck on that display, i.e. continues to show this?  And then does not react any more to any key presses?

PiroRiroPL, 2023-12-13:
Only display its frozen. The web interface is still working, and yes the serial output its going on.
When i click left or middle button the display stock and nothing its changing. 

Here i add my hardware and screen configuration:
[image]
[image]

I have make a video whats show that problem. 
I hope that the quality of the video it's good enough.
https://github.com/dl9rdz/rdz_ttgo_sonde/assets/146319901/7a126ac9-ec48-4287-8522-4ffb6dd73d8b

Now i'm using the newest devel soft.

hr-ru, 2024-01-16:
I so far was unable to reproduce. Maybe one of you could share your config file with me by email (dl9rdz@darc.de)?

You can either get it from the web interface, should be one of the following URLs (replace rdzsonde.local with ip in case mDNS is not working for you)
http://rdzsonde.local/edit.html?file=config.txt
http://rdzsonde.local/file/config.txt
Or use "scripts/ttgoconfig get file config.txt"

hr-ru, 2024-01-16:
Alright, you are both using GPIO21 and GPIO22 for connecting the display. These two pins are used by the I2C connection to the power management controller of the board (on any T-Beam except the very old v0.7).  You cannot use it for a SPI connection to a display at the same time.

I assume this causes problems in particular when pressing buttons (in particular one of the buttons), because the button is connected to the power management controller, and information about button presses is sent via the I2C connection. There is a good chance that this conflicts with display operation. Or, other possibility, when you switch to the display that shows battery voltage. This also communicates with the power management chip via I2C. 

The reason why you might have had fewer problems with older firmware is that the older firmware did not support the new AX2101 power management chip on the T-Beam 1.2 boards.

Fix should be simple: Use different pins for connecting the TFT display. My standard suggestion for T-Beam 1.0/1.1/1.2 is 1/4/13/14/2 for SDA/CLK/RST/RS/CS

See wiki: "you should not connect a SPI display to pin 21/22" (https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Supported-boards)

CT2IRJ, 2024-01-16:
Ok.
Despite being a noob :) on this subject, i understand the reason of conflicting addresses for two separate functions.
I'll change settings and wiring when I get home, and report back.

Thanks again for your insight.
73

dl9rdz, 2024-01-16:
Ok, perfect.

I strongly assume that this also solves the problem of @PiroRiro , as the startup log above similarly shows the use of ports 21 and 22
```
configuration option 'oled_sda'=21 
configuration option 'oled_scl'=22 
```

I am thus closing the issue, feel free to re-open in case it does not solve problem for you.

closed 2024-01-16 by dl9rdz

PiroRiro, 2024-01-29:
I had use the same configuration like @CT2IRJ and now everything works very well :) thx anyway :)