dl9rdz/rdz_ttgo_sonde #412: Add receiver lat/lon/altitude to ChaseMapper

mpkleine, 2023-12-13:
Would it be possible to add the lat/lon/altitude fields of the receiver to the ChaseMapper JSON output?
The receiver has a GPS receiver and the data should be available on each balloon transmission sent via ChaseMapper.

This would provide the appropriate data to mobile/portable external tracking systems and allow for easy distance/direction calculations between the receiver and the balloon.

Szetya, 2023-12-14:
The chasemapper has its own GPS receiver or receives the coordinates of the chaser from the gpsd or on the 55672 port in a separate packet. The balloon telemetry does not include the chaser coordinates (which is why it won't work) Try using the rdzwx-go mobile app it shows distance and direction. https://github.com/dl9rdz/rdzwx-go

mpkleine, 2023-12-14:
OK, thanks, since the receiver has its GPS location I was looking for a way to get it out of the receiver. -- Thanks

Szetya, 2023-12-14:
Another possible solution 
Make HTTP requests to http://rdzsonde-local/live.json (or IP address). The json contains all the useful information, including the GPS or fixed coordinates of the host. I know this works the other way round as you would like.

hr-ru, 2023-12-14:
Chasemapper telemetry can include chase car positions, so the TTGO position could be included there.

But if it is just about getting the position to some other custom (non-chasemapper) application, there are several options, most preferred first:
- Connect to port 14570. You get JSON data every second, which includes the GPS position
- As mentioned above, fetch http://rdzsonde-local/live.json. Very similar data, but you have to pull periodically. For continuously getting the data, the port 14570 is the better option.
- You could also use some of the APRS options (KISS TNC or AXUDP). You will get the GPS position you want (but other data with the limitation of APRS data)

darksidelemm, 2024-01-09:
If you are trying to use the TTGO GPS receiver as a 'car position' input into chasemapper, then the correct format to send data out would be:
```
{
    'type' : 'GPS',            # Packet Type
    'latitude': lat,           # float, Decimal Degrees
    'longitude': lon,          # float, Decimal Degrees
    'altitude': alt,           # float, metres
    'speed': speed,            # float, kph
    'valid': position_valid    # boolean - True if there is GPS lock, False otherwise (probably redundant...)
}
```

If you configure chasemapper to use a car_source type of 'horus_udp', then this will be picked up and used for the car position.

There is an assumption made here that all telemetry data and the car are all at the same location...