dl9rdz/rdz_ttgo_sonde #414: New receiver board to add to the list and feature request

gariac, 2023-12-20:
I got a Lilygo T3-v1.6.-1. The latest code works fine. Regarding sensitivity, it seems to be on par with my T22 V1.1. 

Link:
https://www.lilygo.cc/products/lora3

The desired feature would be to log the received data on the SD card.

rpratt20, 2024-01-15:
I received one of these boards last week. I am so far not successful uploading this program with error below.  The factory PAX counter program was running but after several attempts to load, that program no longer boots on power up indicating something has been written or erased.
esptool.py v2.8
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:6b:ea:24:42:e0
Changing baud rate to 921600
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash...
Compressed 4190208 bytes to 713567...

A fatal error occurred: Timed out waiting for packet header

Trace output is quite long with 199 lines as follows:

Begins:esptool.py v2.8
Serial port /dev/ttyACM0 
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 

break/break

ends:
TRACE +0.000 Received full packet: 010b0400ffffff0000000000
Erasing flash...
Compressed 4190208 bytes to 708025...
TRACE +0.274 command op=0x10 data len=16 wait_response=1 timeout=125.706 data=00f03f00b40200000004000000100000
TRACE +0.000 Write 26 bytes: 
    c000101000000000 0000f03f00b40200 | ...........?....
    0000040000001000 00c0             | ..........
TRACE +125.804 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header

Does anyone know how to upload given this error?

hr-ru, 2024-01-15:
What exact command are you using for flashing? 

If you are using the esptool.py on the command line manually:
I can reproduce your error with that board if using --no-stub (and your output seems to indicate that you are using that option).  It works fine with the stub (i.e. removing the --no-stub option from the command line)

rpratt20, 2024-01-15:
Thank you to  all of you.  I did not have success with esptool using baud rate down to 9600.  However, I changed to a Windows computer and downloaded ESP32_FLASH_DOWNLOAD_TOOL V3.9.5. This worked with baud set to 115200.

gariac, 2024-03-22:
Apologies for not following up on this "issue."  Anyway I just used my normal programming. This is my notes template.

`esptool.py --chip esp32 --port /dev/ttyACM0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 devel20220123-B16-full.bin
` 
Of course I change the bin as required. Other than speed it looks identical to hr-ru.