dl9rdz/rdz_ttgo_sonde #419: Disable other radiosondes decoding?

EricFROL, 2024-01-07:
Hello.
Don't know if theres an option in settings, can't find it... is there any way to just decode M10 sondes?

dl9rdz, 2024-01-07:
Whatever you select in the frequency configuration (QRG tab). 

For each configured frequency, you can select exactly one decoder. No other types will be decoded.

Well, specifically for M10, there is only a single decoder that handles M10 and M20. These are very similar. So if this is your question, then no, there is no way to configure such that just M10 and no M20 will be decoded.

EricFROL, 2024-01-07:
Ok thank you. Didn't see QRG tab at first.

EricFROL, 2024-01-07:
Last question, is this searching for sondes all the time?

closed 2024-01-07 by EricFROL

dl9rdz, 2024-01-07:
The hardware is a single channel receiver (unlike other SDR-based approaches).

So either it is decoding a sonde on one frequency (then it is not searching), or it is searching (cyclic over all configured frequencies)