dl9rdz/rdz_ttgo_sonde #422: Question: Historical data?

EricFROL, 2024-01-14:
Is the data saved anywhere? Something like autorx historical data?

hr-ru, 2024-01-14:
Well, it is not saved on the TTGO, there are no resources for that (maybe in the future some option for storage on SD card might be added - some rudimentary source code is on git for that)

Storage outside depends on what you configure as external destinations where data is sent to :-). 

So that is up to you to set up, either using public servers, with sondehub.org, wettersonde.de, and radiosondy.info being the most popular services, or you can also use any of the many options for self-hosted services, including MQTT (which some people combined with some database backend and grafana), aprsmap, chasemapper, etc.

EricFROL, 2024-01-15:
You know any grafana dashboard reference?

DG0MG, 2024-01-19:
Go to Sondehub.org , enter the probe serial number in the search field, and if data for this probe is found, then click "PLOTS":

[image]

EricFROL, 2024-01-23:
> Go to Sondehub.org , enter the probe serial number in the search field, and if data for this probe is found, then click "PLOTS":
> 
> ![grafanaplot](https://private-user-images.githubusercontent.com/121504683/298200731-a02792fa-e7ce-4541-b16f-2da92d32b587.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwNDgzNDIsIm5iZiI6MTcwNjA0ODA0MiwicGF0aCI6Ii8xMjE1MDQ2ODMvMjk4MjAwNzMxLWEwMjc5MmZhLWU3Y2UtNDU0MS1iMTZmLTJkYTkyZDMyYjU4Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyM1QyMjE0MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYWY4YzlhYzNjYzM0Y2UyMGI2ZWQyYTkyYzcxZGQ5ZDE2MzFhYzEyNmM3ZmMwYmYxMTUyNGU2NTlmOWFkYjY5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2wg5Lg1LVljtw3t7_60ZTKmXZNlejEfFT895voPXceM)

I want to have my own data stored, not sondehub

hr-ru, 2024-01-23:
Unfortunately I don't have any good references.  I just had heard it from a few stations that they are using Grafana in some way. One example was in a discussion here, see https://github.com/dl9rdz/rdz_ttgo_sonde/issues/45

bretberger, 2024-03-06:
Along the same lines I am wondering about access to the received data telemetry stream via serial port on the TTGO. My potential use case would be a TTGO board connected to a large battery installed at a remote and favorable mountain top location for receiving location data for landed sondes in my area. Telemetry would be parsed and stored by an external uP board such as Raspberry Pi Pico. Once a month, or so, landed sonde data would be retrieved and battery replaced/recharged. Would prefer, for power consumption reasons, to keep WiFi powered off.

dl9rdz, 2024-03-06:
The serial port is currently used for logging debug data. You actually will find the telemetry in there if you want to (if you enable the right data feeds.
- the chasemapper connector should log all messages it sends out (simple json) (just the basic data)
- the mqtt connector should log all messages it attempts to sent to a mqtt server (more data, should work even if not connected to mqtt server)
- the sondehub connector should log all messages it sends out (more difficult to parse, these are chunked http requests merging multiple frames) - but only if connected to sondehub
- for axupd it is currently commented out in the source (advantage of axudp is that it is sent for all frames)

Currently there is not guarantee that the output will be the same on future versions.

bretberger, 2024-03-06:
Perfect! Physically on the TTGO LORA v2.1_1.6 boards this would be the USB port? Thanks!

dl9rdz, 2024-09-04:
In addition to what was mentioned, the current version also support storing received data locally on a SD card. I fpower consumption is an issue, then this might be better than an external board connected to the Serial port.

closed 2024-09-04 by dl9rdz