dl9rdz/rdz_ttgo_sonde #423: Monitor battery voltage from web interface

BadCactus634, 2024-01-15:
Hi, sorry if this is not the right way to post my question but I can't find any info nor community forum online.
I have a TTGO LoRa32 V2.1_1.6 with the version  0.9.3 installed. I connected the board to a 5V regulated solar panel and put it on my roof. The board is connected to my WiFi.
The problem is, I can't have access to the battery info in any way through the web interface, and this is frustrating since I don't have any way to know if the battery is dying of it is being recharged by the solar panel via the 5V micro-USB input.

Is there any way (maybe a hidden link) to access the measurement that the board takes from the voltage divider on pin 35 besides the display?
I think that adding a voltage indicator on the top-right of the web interface, in the menu bar, would be awesome.

Thank you!

hr-ru, 2024-01-15:
Not at the moment, but there are plans to add a status page to the web interface. This likely will include board info (uptime, battery status) as well as connection status (e.g. errors from sondehub connection).

BadCactus634, 2024-01-16:
I hope this will be implemented soon. Thanks!

closed 2024-01-16 by BadCactus634