dl9rdz/rdz_ttgo_sonde #426: Sondehub: same position after gap

rs1729, 2024-02-06:
devel20231110

After a gap of 20 minutes the last position/velocity was recycled and resent to sondehub.
frame 2101:
```
    {
        "software_name": "rdzTTGOsonde",
        "software_version": "devel20231110",
        "uploader_callsign": "SQxxx",
        "time_received": "2024-02-05T16:49:01.000Z",
        "manufacturer": "Vaisala",
        "serial": "V1010258",
        "datetime": "2024-02-05T16:49:18.000Z",
        "lat": 52.20412,
        "lon": 14.19055,
        "alt": 1424.09473,
        "frequency": 405.1,
        "vel_h": 27.88741,
        "vel_v": 4.97614,
        "heading": 106.14401,
        "rssi": -107.5,
        "frame": 2101,
        "type": "RS41",
        "sats": 8,
        "batt": 2.9,
        "position": "52.20412,14.19055",
    },
```
frame 3462:
```
    {
        "software_name": "rdzTTGOsonde",
        "software_version": "devel20231110",
        "uploader_callsign": "SQxxx",
        "time_received": "2024-02-05T17:11:42.000Z",
        "manufacturer": "Vaisala",
        "serial": "V1010258",
        "datetime": "2024-02-05T17:11:59.000Z",
        "lat": 52.20412,
        "lon": 14.19055,
        "alt": 1424.09473,
        "frequency": 405.1,
        "vel_h": 27.88741,
        "vel_v": 4.97614,
        "heading": 106.14401,
        "rssi": -106.5,
        "frame": 3462,
        "type": "RS41",
        "sats": 8,
        "batt": 2.8,
        "position": "52.20412,14.19055",
    },
```

rs1729, 2024-02-10:
Perhaps similar problem: V3750539
https://testing.v2.sondehub.org/#!mt=Mapnik&mz=12&qm=3h&mc=51.16557,5.64354&f=V3750539&q=V3750539

cf.
https://github.com/dl9rdz/rdz_ttgo_sonde/issues/390
If there is no valid position, the receiver should not upload a new frame with old GPS data to Sondehub as different receivers might have different last positions.

KyotoFox, 2024-07-16:
I also encountered this issue today with [V3050409](https://sondehub.org/#!mt=Mapnik&mz=10&qm=12h&mc=59.1009,5.83099&f=V3050409&q=V3050409)

It doesn't really look like a GPS issue though. Looking at the [SondeHub logs](https://grafana.v2.sondehub.org/d/bbaa7894-e5f4-4c0d-be96-897b4ffde43b/radiosonde-telemetry-dashboard?var-Serial=V3050409&from=1721084634000&to=1721093626000&orgId=1), there are lots of receivers that receive frame 8414 & 8416, but only my `KYOTO-RDZTTGO` report frame 8415, where it recycles the position & altitude it last received (frame 4426)

The frames in question from my rdzTTGO:
```
4426 {"software_name":"rdzTTGOsonde","software_version":"master_v0.9.4","uploader_callsign":"KYOTO-RDZTTGO","time_received":"2024-07-15T23:52:54.000Z","manufacturer":"Vaisala","serial":"V3050409","datetime":"2024-07-15T23:53:10.000Z","lat":59.06171,"lon":6.18413,"alt":11582.4707,"frequency":403,"vel_h":15.18862,"vel_v":4.15337,"heading":44.72972,"rssi":-116.5,"frame":4426,"type":"RS41","sats":8,"subtype":"RS41-SGP","temp":-44.3,"humidity":1.8,"pressure":212.78,"batt":2.7,"user-agent":"Amazon CloudFront","position":"59.06171,6.18413","upload_time_delta":-19.15}

8415 {"software_name":"rdzTTGOsonde","software_version":"master_v0.9.4","uploader_callsign":"KYOTO-RDZTTGO","time_received":"2024-07-16T00:59:23.000Z","manufacturer":"Vaisala","serial":"V3050409","datetime":"2024-07-16T00:59:39.000Z","lat":59.06171,"lon":6.18413,"alt":11582.4707,"frequency":403,"vel_h":15.18862,"vel_v":4.15337,"heading":44.72972,"rssi":-117,"frame":8415,"type":"RS41","sats":8,"subtype":"RS41-SGP","temp":-35.6,"humidity":0.3,"pressure":11.26,"burst_timer":65535,"batt":2.6,"user-agent":"Amazon CloudFront","position":"59.06171,6.18413","upload_time_delta":-19.934}
8429 {"software_name":"rdzTTGOsonde","software_version":"master_v0.9.4","uploader_callsign":"KYOTO-RDZTTGO","time_received":"2024-07-16T00:59:37.000Z","manufacturer":"Vaisala","serial":"V3050409","datetime":"2024-07-16T00:59:53.000Z","lat":59.22508,"lon":5.9708,"alt":31244.63867,"frequency":403,"vel_h":13.16257,"vel_v":6.39631,"heading":282.9946,"rssi":-118,"frame":8429,"type":"RS41","sats":10,"subtype":"RS41-SGP","temp":-35.6,"humidity":0.3,"pressure":11.17,"burst_timer":65535,"batt":2.6,"user-agent":"Amazon CloudFront","position":"59.22508,5.9708","upload_time_delta":-19.934}
8447 {"software_name":"rdzTTGOsonde","software_version":"master_v0.9.4","uploader_callsign":"KYOTO-RDZTTGO","time_received":"2024-07-16T00:59:55.000Z","manufacturer":"Vaisala","serial":"V3050409","datetime":"2024-07-16T01:00:11.000Z","lat":59.22505,"lon":5.96742,"alt":31338.27734,"frequency":403,"vel_h":10.52642,"vel_v":5.64434,"heading":260.27802,"rssi":-117.5,"frame":8447,"type":"RS41","sats":10,"subtype":"RS41-SGP","temp":-35.6,"humidity":0.3,"pressure":11.01,"burst_timer":65535,"batt":2.6,"user-agent":"Amazon CloudFront","position":"59.22505,5.96742","upload_time_delta":-19.263}
```

And including the frames before/after (from other receivers):
```
8414 {"software_name":"radiosonde_auto_rx","software_version":"1.7.2","uploader_callsign":"Megafluffy_AUTO_RX","uploader_position":"60.32,5.29","uploader_antenna":"J-Pole","time_received":"2024-07-16T00:59:23.131682Z","datetime":"2024-07-16T00:59:38.999000Z","manufacturer":"Vaisala","type":"RS41","serial":"V3050409","subtype":"RS41-SGP","frame":8414,"lat":59.225,"lon":5.97404,"alt":31163.28821,"temp":-35.7,"humidity":0.3,"pressure":11.3,"vel_v":5.18223,"vel_h":12.95394,"heading":286.63554,"sats":10,"batt":2.6,"frequency":403.001,"burst_timer":65535,"ref_position":"GPS","ref_datetime":"GPS","rs41_mainboard":"RSM421","rs41_mainboard_fw":"20215","snr":22.3,"tx_frequency":403,"user-agent":"Amazon CloudFront","position":"59.225,5.97404","upload_time_delta":-0.873,"uploader_alt":80}
8415 {"software_name":"rdzTTGOsonde","software_version":"master_v0.9.4","uploader_callsign":"KYOTO-RDZTTGO","time_received":"2024-07-16T00:59:23.000Z","manufacturer":"Vaisala","serial":"V3050409","datetime":"2024-07-16T00:59:39.000Z","lat":59.06171,"lon":6.18413,"alt":11582.4707,"frequency":403,"vel_h":15.18862,"vel_v":4.15337,"heading":44.72972,"rssi":-117,"frame":8415,"type":"RS41","sats":8,"subtype":"RS41-SGP","temp":-35.6,"humidity":0.3,"pressure":11.26,"burst_timer":65535,"batt":2.6,"user-agent":"Amazon CloudFront","position":"59.06171,6.18413","upload_time_delta":-19.934}
8416 {"software_name":"radiosonde_auto_rx","software_version":"1.7.2","uploader_callsign":"Megafluffy_AUTO_RX","uploader_position":"60.32,5.29","uploader_antenna":"J-Pole","time_received":"2024-07-16T00:59:25.139012Z","datetime":"2024-07-16T00:59:40.999000Z","manufacturer":"Vaisala","type":"RS41","serial":"V3050409","subtype":"RS41-SGP","frame":8416,"lat":59.22504,"lon":5.9736,"alt":31174.10582,"temp":-35.6,"humidity":0.3,"pressure":11.25,"vel_v":4.73224,"vel_h":12.84332,"heading":273.87529,"sats":10,"batt":2.6,"frequency":403.001,"burst_timer":65535,"ref_position":"GPS","ref_datetime":"GPS","rs41_mainboard":"RSM421","rs41_mainboard_fw":"20215","snr":21.4,"tx_frequency":403,"user-agent":"Amazon CloudFront","position":"59.22504,5.9736","upload_time_delta":-0.873,"uploader_alt":80}
```

I'll try to dig a bit around in the code to see where this occurring later!

dl9rdz, 2024-07-16:
This should be fixed already in commit  20a5d728051ec96f3c47bd2493fe873d18149a88 (Feb 20).

However, this is only in the devel version, seems like you are using the master version from January. So I guess I really should release a new master version soon....

KyotoFox, 2024-07-16:
Ah, perfect! I'll make sure to upgrade my board as well then. 
I'm already running `devel20240521` on another board, but that one didn't have reception at the time

closed 2024-09-04 by dl9rdz