dl9rdz/rdz_ttgo_sonde #428: MTS01 decode

alkolbender, 2024-03-03:
Hi 
My QTH turkey /istanbul 

im decode rpi_zero and radiosonde_auto_rx decode signal but rdz_ttgo_sonde dont 

Can you help me?

i trycchange offset but not work . 
radiosonde_auto_rx  scan.py = Meteosis MTS01 Sonde  Score: %.2f, Offset: %.1f Hz
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/scan.py

Is there a way to decode the MTS01 signal with ttgo?

dl9rdz, 2024-03-03:
MTS01 decoding is not supported currently.

It seems to be transmitting in AFSK (after briefly looking at auto_rx source code), which is not feasible to decode with the SX127x chips on the TTGO

alkolbender, 2024-03-03:
thanks for the quick reply

I will wait for my relatives to pass DFM17.

happy hunting

73.