dl9rdz/rdz_ttgo_sonde #431: Bug T-BEAM 1.2 - AXP2101

14ri004, 2024-03-23:
hello, I received a T-BEAM 1.2 - AXP2101 (instead of an AXP192), I installed RDZ but the T-BEAM has a strange behavior, scan that panics, no probe decoding even at 1m distance, crashes, screen freezes, no recharging of the battery, no display of battery voltage on the screen, finally a lot of problems that I don't have with my old T-BEAM AXP192, what solution??, thank you

dl9rdz, 2024-03-23:
APX2101 should not be a problem (some minor differences, like charging current cannot be displayed), but there are two versions available, one with SX1262 and one with SX1278 chip.  

The SX1262 does not work (you probably can recognize this if the silver TTGO sticker says "SX1262" in the last (4th) line), the SX1278 should work (for that the sticker just says LILYGO|433/470MHz|Modle:TTGO). Which if the two do you have?

Good: [image]
Bad: [image]

14ri004, 2024-03-23:
I have version v1.2 SX1262 430/470, this is not the version I ordered, but they sent me this one because the version with the AXP192 chip is no longer marketed, I am returning it and refund, I ordered another T-BEAM in another store.

dl9rdz, 2024-03-24:
I have good experience with directly ordering from Lilygo (e.g. https://www.lilygo.cc/products/t-beam-softrf?variant=43170139930805). There you can chose which version (nowadays, you always get V1.2 with AXP2101, but you can choose between the SX1262 ans SX1278 version...) and they usually deliver what you order...

closed 2024-04-07 by 14ri004