dl9rdz/rdz_ttgo_sonde #433: RS41 Humidity calculation

srcejon, 2024-04-09:
In RX_FSK/src/RS41.cpp, there's code based on https://github.com/einergehtnochrein/ra-firmware

Around line 648 the code is:

       float Tp = (sensorTemp - 20.0f) / 180.0f;
       float sum = 0;
       float powc = 1.0f;
       float p = pressure / 1000.0f;
       for ( int i = 0; i < 3; i++) {
          float l = 0;
          float powt = 1.0f;
          for ( int j = 0; j < 4; j++) {
             l += calibration->value.matrixBt[4*i+j] * powt;
             powt *= Tp;
          }
          float x = calibration->value.vectorBp[i];
          sum += l * (x * p / (1.0f + x * p) - x * powc / (1.0f + x));
          powc *= Cp;
       }
       Cp -= sum;

       // Should sum be set to 0 here?

       float xj = 1.0f;
       for ( int j = 0; j < 7; j++) {
          float yk = 1.0f;
          for ( int k = 0; k < 6; k++) {
             sum += xj * yk * calibration->value.matrixU[j][k];
             yk *= Tp;
          }
          xj *= Cp;
       } 

       float RH = sum

However, in the original version of the code https://github.com/einergehtnochrein/ra-firmware/blob/master/src/rs41/rs41metrology.c at line 271, sum is set to 0 between the two loops, but that seems to be missing here.

mycarda, 2024-06-19:
I think you are correct!

I have not looked at this code for almost three years. In the first ```for``` loop (line 652) a polynomial correction is being calculated for the (already compensated) capacitance value ```Cp```  from the humidity sensor.  ```sum``` is used to sum up the polynomial terms and the correction is applied on line 663. At this point, ```sum``` should be set to zero as it is used again in the next ```for``` loop where the corrected capacitance ```Cp``` is used to calculate the relative humidity ```RH```. @srcejon you obviously already knew that because you raised the issue.

The interesting question is why the current code gives anything like a realistic value for the relative humidity. Usually, forgetting to zero a variable would have disastrous affects. I looked at some typical values received from an RS41 and the correction applied to ```Cp``` at line 633 is usually quite small (0.002). So, I think the answer to this question is that ```sum``` is usually small at line 633 so setting it to zero does not make a lot of difference. However, this is not an excuse. I should have set it to zero three years ago. :-) 

I will go find a spare TTGO Lora board in my box of bits and pull the latest code then make the trivial fix. It it all tests OK, I will submit the update.

mycarda, 2024-07-01:
Tested today and created pull request [445](https://github.com/dl9rdz/rdz_ttgo_sonde/pull/445).

It took far longer than expected to test this because I kept getting unrelated TG1WDT_SYS_RESET task timer problems. I had to comment out the u8x8->begin() in display.cpp just to stop it resetting every second. At first, I thought it was a problem with my TTGO hardware but I loaded a very old copy of the software from three years ago and the hardware was fine.

On the test today, there was very little difference to the relative humidity value. I tracked it for a range of relative humidity results and the following is what I logged (for the very few of those who are interested). CpSum is the value of sum between the two relative humidity corrections. HR is the calculation with sum set to zero and old HR is the previous result without sum set to zero. As you can see, there is very little difference.

HUMIDITY TEST: CpSum 0.015037, sum 1.643079, RH 2.835819, oldRH 2.861772
HUMIDITY TEST: CpSum 0.015044, sum 1.673621, RH 2.877350, oldRH 2.903214
HUMIDITY TEST: CpSum 0.015048, sum 1.691820, RH 2.896165, oldRH 2.921924
HUMIDITY TEST: CpSum 0.015046, sum 1.692891, RH 2.885373, oldRH 2.911018
HUMIDITY TEST: CpSum 0.015040, sum 1.697369, RH 2.882629, oldRH 2.908171
...
HUMIDITY TEST: CpSum 0.015307, sum 3.394660, RH 6.242510, oldRH 6.270659
HUMIDITY TEST: CpSum 0.015369, sum 3.531196, RH 6.721958, oldRH 6.751214
HUMIDITY TEST: CpSum 0.015449, sum 3.754591, RH 7.335121, oldRH 7.365302
HUMIDITY TEST: CpSum 0.015509, sum 3.951676, RH 7.787828, oldRH 7.818393
HUMIDITY TEST: CpSum 0.015567, sum 4.185463, RH 8.313091, oldRH 8.344011
...
HUMIDITY TEST: CpSum 0.016529, sum 11.449142, RH 21.094282, oldRH 21.124735
HUMIDITY TEST: CpSum 0.016530, sum 11.581532, RH 21.272604, oldRH 21.302965
HUMIDITY TEST: CpSum 0.016539, sum 11.721081, RH 21.358971, oldRH 21.389111
HUMIDITY TEST: CpSum 0.016544, sum 11.873607, RH 21.566071, oldRH 21.596119
...
HUMIDITY TEST: CpSum 0.015759, sum 39.392155, RH 66.296585, oldRH 66.323105
HUMIDITY TEST: CpSum 0.015595, sum 38.235714, RH 64.264565, oldRH 64.290771
HUMIDITY TEST: CpSum 0.015515, sum 38.557415, RH 64.500595, oldRH 64.526543
...

dl9rdz, 2024-07-01:
I already merged this (to the devel branch... will propagate to the main branch when I create the next version)
However, the automated builds are currently not working, and I am travelling this week, so will take a few days for a new binary image to become available.